raspberrypi/noobs #223: BUILDROOT_DL_DIR not working

procount, 2015-01-16:
The value of BUILDROOT_DL_DIR set in BUILDME.sh is not percolating down to rpi-firmware,mk and rpi-userland.mk.
Maybe need to add export in BUILDME.sh???

lurch, 2015-01-16:
The intention isn't for `BUILDROOT_DL_DIR` to be 'set' in `BUILDME.sh`, but merely for `BUILDME.sh` to use the correct value (for reading the rpi-\*-head.version files) **if** it's been set as a global environment variable, as described in http://buildroot.uclibc.org/downloads/manual/manual.html#env-vars
(That link describes `BR2_DL_DIR`, but in the old version of Buildroot used by NOOBS it's still named `BUILDROOT_DL_DIR`).

So you *could* do something like:
```
mkdir $HOME/buildroot_downloads
export BUILDROOT_DL_DIR=$HOME/buildroot_downloads
./BUILDME.sh
```
and then if you had multiple separate NOOBS / Buildroot trees (as I do) it'd simply store all the downloaded archives in `$HOME/buildroot_downloads` rather than having separate (duplicated) `dl` folders inside each `buildroot` directory.

Or if the `BUILDROOT_DL_DIR` env var isn't set, it'll just default to `dl` (it's relative to the `buildroot` directory) as per normal.

procount, 2015-01-16:
OK, but I'm just saying that if the average joe (me) downloads the
latest noobs buildroot and executes ./BUILDME.sh without doing anything
else, then rpi-userland.mk and rpi-firmware.mk WILL fail because
BULDROOT_DL_DIR is blank or non-existent at that point. So probably in
your environment you have it set globally, or something, but the default
override of setting it to "dl" is not working for me on a clean ubuntu
environment when the env_var does not exist. 

Regards, 

Kevin 

On

> The intention isn't for
BUILDROOT_DL_DIR to be 'set' in BUILDME.sh, but merely for BUILDME.sh to
use the correct value (for reading the rpi-*-head.version files) IF it's
been set as a global environment variable, as described in
http://buildroot.uclibc.org/downloads/manual/manual.html#env-vars [1]
(That link describes BR2_DL_DIR, but in the old version of Buildroot
used by NOOBS it's still named BUILDROOT_DL_DIR). 
> 
> So you _could_
do something like: 
> 
> mkdir $HOME/buildroot_downloads export
BUILDROOT_DL_DIR=$HOME/buildroot_downloads ./BUILDME.sh 
> 
> and then
if you had multiple separate NOOBS / Buildroot trees (as I do) it'd
simply store all the downloaded archives in $HOME/buildroot_downloads
rather than having separate (duplicated) dl folders inside each
buildroot directory. 
> 
> Or if the BUILDROOT_DL_DIR env var isn't set,
it'll just default to dl (it's relative to the buildroot directory) as
per normal. 
[2].

 

Links:
------
[1]
http://buildroot.uclibc.org/downloads/manual/manual.html#env-vars
[2]
https://github.com/raspberrypi/noobs/issues/223#issuecomment-70258323

lurch, 2015-01-16:
Oh, whoops :-(

I thought I'd tested it and found that if `BUILDROOT_DL_DIR` wasn't set, then Buildroot 'internally' set it to `dl` just for the duration of the make?
But perhaps I didn't test thoroughly enough (I guess I should have done a test in a clean-install VM just to be sure), so I'll try to get this working properly over the weekend.

I guess in the meantime it should be fairly easy to work around this problem (if you haven't already done so!) by setting the global env-var.

procount, 2015-01-16:
Well, I understand how it is supposed to work now, it just didn't
work for me straight out of the box. My solution was to just export it
from BUILDME.sh and it then worked fine. 

As I have just done a
complete build from scratch of your latest github version on a new VM of
Ubuntu 14.04, There are still 2 additional outstanding issues which
prevent it from building out of the box and I think they should be
pretty easy to fix, even if you have a more longer term solution in mind
for either of them. 

#158 - add libqt4-dev to the list of dependencies
to be installed. 

#171 (last comment) - upgrade wget from 1.14 to 1.15
or 1.16 since it won't compile. (some error with a .pod file) 

At least
with these 3 issues fixed it should then build straight out of the box
(in Ubuntu 14.04 anyway) 

Hope this helps, 

Regards, 

Kevin. 

On

> Oh, whoops :-( 
> 
> I
thought I'd tested it and found that if BUILDROOT_DL_DIR wasn't set,
then Buildroot 'internally' set it to dl just for the duration of the
make?
> But perhaps I didn't test thoroughly enough (I guess I should
have done a test in a clean-install VM just to be sure), so I'll try to
get this working properly over the weekend. 
> 
> I guess in the
meantime it should be fairly easy to work around this problem (if you
haven't already done so!) by setting the global env-var. 

Links:
------
[1]
https://github.com/raspberrypi/noobs/issues/223#issuecomment-70262723

commit 1539d369 (2015-02-07): Use explicit package versions

This makes the builds more reprodcible, removes the need for all
the 'version hackery', and also fixes #223
closed 2015-02-07 by lurch