raspberrypi/noobs #224: missing "config.txt" template file in NOOBS boot partition

beta-tester, 2015-01-19:
related to this thread http://www.raspberrypi.org/forums/viewtopic.php?f=28&t=96853

i think, it would be helpful to others, to add a default config.txt template file to the NOOBS boot partition, so that everybody can intuitively see, that there is a config.txt that they can tweak to their own needs.

and to add a note to NOOBS that also there a config.txt can be used to tweak the NOOBS system.

it would be also nice to have the ability in NOOBS to create/edit that config.txt for NOOBS itself as well as these for the other installed OS'es.

lurch, 2015-02-08:
Not sure about this - 99% of people won't need a config.txt file for NOOBS itself, so adding a 'template config.txt' would only serve to confuse the majority of people (they'd get confused between the config.txt used by NOOBS, and the config.txt used by the OS).

As a side-note I agree the display-mode-selection of NOOBS currently has a few 'issues', so there are plans (with no ETA yet I'm afraid) to give this a bit of an overhaul.

beta-tester, 2015-02-10:
hmmm... yes, maybe it will confuse  these 99% of people.
but many people are already confused because of the config.txt for the installed OS'es,
where to find them on the SD card when it is plugged to a windows PC...
it is not well documented, that each installed OS has its own boot partition, that is possibly not visible under windows...
many people do not know - for unknown reason - that they have the ablility to adjust the files already in NOOBS own editor... but they sould also have an option to adjust/create the config.txt and cmdline.txt for NOOBS itself... for example the cmdline option `usbhid.mousepoll=0` is needed for my mouse, otherwise it is terrible to work with my mouse in NOOBS.

lurch, 2015-02-10:
> many people do not know - for unknown reason - that they have the ablility to adjust the files already to edit OS's config in NOOBS own editor...

Suggestions for how the NOOBS documentation could be improved are always welcome... :-)

> an option to adjust/create the config.txt and cmdline.txt for NOOBS itself...
> for example the cmdline option
> `usbhid.mousepoll=0`
> is needed for my mouse, otherwise it is terrible to work with my mouse in NOOBS.

Do you have this problem only with older versions of NOOBS, or with the latest v1.3.12 as well? (with the updated linux kernel)
As https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained#noobs-bootup-low-level sort of explains, under NOOBS the `cmdline.txt` file is renamed to `recovery.cmdline`

beta-tester, 2015-02-16:
Yes, with NOOBS 1.3.12 i still have to add `usbhid.mousepoll=0` to `recovery.cmdline`, because of the extreme slow mouse movement with very large latency.

beta-tester, 2015-02-19:
and same slow mouse behaviour with NOOBS 1.4 !
why not add usbhid.mousepoll=0 to the recovery.cmdline by default, to be sure every one has a good mouse handling... even if the user has such s silly mouse that need that option...
othere mices will not be affected by this option... and it is only for NOOBS... for all other cmdline.txt files the user has to care about, to add that option in case if a silly mouse.

lurch, 2015-02-19:
I'm afraid I know nothing about this mousepoll cmdline option (and what side-affects it may have).
Any opinions from @ghollingworth @popcornmix or @asb ?

procount, 2015-02-19:
I also like this feature, but for different reasons. On the HDMIPI, I find it a lot more readable to drop the resolution down to its native 1280x800 (especially when dropping into the other terminal for debugging).
I am creating a branch of noobs that will include this feature to edit config.txt and recovery.cmdline for the NOOBS partition. I'll submit a Pull Request when it is complete.

procount, 2017-11-29:
PINN now supports editing of the recovery.cmdline and config.txt file in the recovery partition.

XECDesign, 2017-11-29:
Closing as a wontfix here then with the suggestion that if people require more than a simple installer and boot manager they should switch to NOOBS. Thanks, @procount

closed 2017-11-29 by XECDesign