raspberrypi/noobs #225: cat: dl/rpi-userland-head.version: No such file or directory

vermicelli, 2015-01-20:
the "rpi-userland-head.version" and the "rpi-firmware-head.version" are used in the BUILDME.sh(line 52,53). but i can't find these files in directory buildroot/dl

procount, 2015-01-20:
They are created by rpi-userland.mk#16 and rpi-firmware.mk#14 respectively, so you need to be online and have git installed in order for this information to be udpated.

lurch, 2015-01-20:
I wonder if this is related to #223 and is something else I broke in the same commit?

Super-busy at mo, but I'll try to get it fixed by the weekend.

procount, 2015-01-20:
Ahh yes! I seem to recall I had a blank BUILD-DATA file before I exported BUILDROOT_DL_DIR from BUILDME.sh. @vermicelli - please try my temporary fix mentioned in #223 to see if that works, until @lurch has time to provide a proper fix.

lurch, 2015-01-20:
Closing as duplicate of #223

closed 2015-01-20 by lurch