raspberrypi/noobs #226: Noobs 1.4 for Pi2

Mestapholes, 2015-02-03:
Hello,

I ordered a couple of Pi2s for some colleagues and me.
What I forgot was to order sd cards with the noob software.
On the rs components site it states that these cards have a noob 1.4 version.
But I can not find this version in the download (latest is 1.3.12) section or here.
Does this old version run on the new Pi or would there be issues?
Thanks in advance
Mestapholes

Mestapholes, 2015-02-03:
Sorry. I just notices myself that the sd card offered by rs components is sceduled for 01.04.
My colleague pointed it out to me that this version was available and told me just now that it is on preorder...

closed 2015-02-03 by Mestapholes

lurch, 2015-02-03:
Unfortunately RS Components are giving out wrong information - NOOBS v1.3.12 is the updated version that supports the Pi2 :-)

ghollingworth, 2015-02-04:
'Just so people get an update'

Yes 1.3.12 is the initial software release but we know there are a couple of things that were going to get updated quickly (Mathematica amongst them) so 1.4 will be released at the beginning of next week