raspberrypi/noobs #230: Install fails on many SD cards with RPi2

digitalLumberjack, 2015-02-09:
Hi,
I cannot install any os from noobs 1.3.12 on some sdcards !
The install sceen is stuck on "Clearing existing EBR" and then fails. 

My sdcard is a noname 8gb.

dmesg specify the following : (sorry for the picture but i couldnt mount a partition to write dmesg output)

[image]

I'm installing on rpi2, all is ok with the same card and same noob version on rpi1.

lurch, 2015-02-09:
Hmmm, strange that this should be specific to the rpi2 and work fine on the rpi1. I never saw any errors like this myself during testing. Any ideas, @ghollingworth , @P33M or @popcornmix ?
The "Clearing existing EBR" message is [here](https://github.com/raspberrypi/noobs/blob/f8a6f302f79fb1641860bd297c917722f825e4dd/recovery/multiimagewritethread.cpp#L115) which then calls [this](https://github.com/raspberrypi/noobs/blob/f8a6f302f79fb1641860bd297c917722f825e4dd/recovery/multiimagewritethread.cpp#L393).

@digitalLumberjack Does this always happen every time on the rpi2? Have you used the official SDFormatter tool with format size adjustment on? What power-supply are you using?
You say "fails on many SD cards" - are they all "noname 8gb" cards? (which obviously makes the problem hard to reproduce)

digitalLumberjack, 2015-02-10:
It's always happen with the pi2, I used the SD formatter on macosx that always worked with any card and os. I'm using a 2AMP 5V power supply. I have many 32G and 16G cards (samsung, transcend lexxard) and i tried all. Everything ok. 
It's just the 8G card. I'm currently getting card models that doesn't work, i'll be reporting this today.

digitalLumberjack, 2015-02-10:
Hi, the other card is a "8GB SanDisk Class 2 card"

lurch, 2015-02-21:
Does NOOBS v1.4.0 also suffer from the same problem?

NicoHood, 2015-02-21:
I am using a Transcend 32gb micro sd (45mb/s) and had the same problem with the old noobs. havent dared to try the newer noobs.
Just wanted to link the forum post: 
http://www.raspberrypi.org/forums/viewtopic.php?p=694741#p694741

It seems that using a different sd card (talking about raspbian only), using rpi-update and transfering it helped. So if 1.4 uses a new kernel this could have solved the problem. But never tested it with noobs.

lurch, 2015-02-21:
Thanks for the link. The "bug in the dynamic clock scaling code" was fixed in the [firmware](https://github.com/raspberrypi/firmware/commit/89efbd45ef316e620749f7007014067257f465af) and NOOBS v1.4 includes [updated firmware](https://github.com/raspberrypi/noobs/releases/tag/v1.4) so this bug should be hopefully fixed now.

So if @digitalLumberjack can test, confirm and close this bug, that would be great :)

NicoHood, 2015-02-21:
Well and today my sd card crashed again with the new firmware. I have to try the "broken" sd card with the old b+ pi and the "working" sd card with the pi2.

crispcbk, 2015-03-12:
I'm trying to start NOOBS v1.4.0 on a SanDisk 8 GB SDHC card on a RPi2. Unfortunately, after selecting the components to install (Raspbian + "include data partition), the setup process stops with the error message "mkfs.fat: warning - lowercase labels might not work properly with DOS or Windows". This is shown in a popup window; after closing the window, the RPi2 reboots and (obviously) the card is not usable.

lurch, 2015-03-12:
@crispcbk Wow, I've never seen that error message before! (and I've obviously done **lots** of testing)
Did you use the SDFormatter tool first? Can you confirm the contents of your `BUILD-DATA` file?

(perhaps this might be related to #140 which I only fixed in development versions of NOOBS and not in the release version?)

crispcbk, 2015-03-12:
Hi lurch, thanks for the quick feedback. I'm a total n00b with respect to RPi2 so please don't hesitate to point out if I'm not providing essential information :).

I tested again on the same card, this time trying to install without creating an additional data partition. Same issue, I've attached a screenshot of the error message

[image]

The BUILD-DATA file contains the following information (not quite sure what to make of it): 
"Build-date: 2015-02-18
NOOBS Version: v1.4
NOOBS Git HEAD @ be1862829eabda6c88856348920d545827b1e82b
rpi-userland Git master @ c5341f09dc686c17966369485f7dd27f58dc081c
rpi-firmware Git master @ 8aca5762f984f6decbeda294cee8418966c3d8d3
rpi-linux Git rpi-3.18.y @ b630db9dcb5c73367050a672823047b0466b16b9"

Let me know if I can provide additional info or test something to continue troubleshooting.

lurch, 2015-03-12:
Thanks for the screenshot :-) It shows that the "lowercase labels ..." message is just a harmless warning that can be safely ignored. The **real** error is "failed whilst writing reserved sector".
If even the official SDFormatter has 'problems' with the card, then it indeed might be that the card is somehow 'broken'? Perhaps the card has been used so much that the wear-levelling has worn out?
I don't think it would be 'incompatible' as the Pi itself has been around for 3 years (as already mentioned, this is the first time I've ever seen that particular error).

The good news is that SD cards are *much* cheaper now than they were 3 years ago... ;)

crispcbk, 2015-03-12:
Hrmmm... had no issues with the card thusfar. Discovered that it is a "class 2" SD card so it might not be as fit for booting NOOBS as I had hoped :). Ordered a Samsung Pro 32 GB card (class 10) and will retry with that one.

spatieman, 2015-03-19:
we tested several SD cards, seems sandisk stuff works best here.
got several time with various card that after a reboot we got masive midle finger work on (re)boot.
we play with the config settings, inc did a firmware upgrade.
looks like high speed card give some isues with it.

lurch, 2015-03-19:
@spatieman Can you please provide exact details / part numbers for the SD cards that aren't working?

NicoHood, 2015-03-19:
http://www.amazon.de/Transcend-Premium-microSDHC-Speicherkarte-SD-Adapter/dp/B00APCMMDG/ref=sr_1_3?ie=UTF8&qid=1426786143&sr=8-3&keywords=transcend+micro+sd+32gb

TS32GUSDU1 does not work for me. After some time it crashes or even after the setup. Could also be related to the OS itself and not noobs since the noobs bootmenu at least always shows up and boots the os.

Could really be a problem of high speed cause this one is really fast. Maybe the OS/Kernel developer should have a look at this.

lurch, 2015-03-19:
Regarding Transcend cards, see https://github.com/raspberrypi/firmware/issues/397 and https://github.com/raspberrypi/firmware/issues/372

sidey79, 2015-03-31:
Hi,

I've a very similar problem with ADATA AUSDH16GUICL10-RA1. Already replaced the card. Same problem.

The PI2 is working till a few reboots. Then i get similar read errors as digitalLumberjack. Now i'am using a older class 4 SD card and having no problem, Seems to be a problem of the pi 2.
I hope it's software related and not hardware.

trueimage, 2015-04-08:
I'm having this problem with a samsung 16gb evo (orange) card. Anything I can do to help? Alternatively anyone have a recommended card that does not have these issues?

Dygear, 2015-05-04:
Yeah, I bought some [32GB Class 10 Cards](http://www.newegg.com/Product/Product.aspx?Item=N82E16820231577) and I'm having the same issue. Cards are very, very fast when formatting (In the 45MB/s range) and copying the NOOBS install, but once I go through the install process it hangs for a bit on the clear EBR thing, and then gives the screen shot shown above. The cards themselves seem to be ok, it's only when used with a Raspberry Pi v2 that seems the cause an issue. I've yet to try this on the Rpi1.

lurch, 2015-05-04:
From the investigation being done on https://github.com/raspberrypi/firmware/issues/397 it sounds like this is definitely specific to the Pi2, and steady progress is being made towards finding a solution.
So this problem should be fixed by the time NOOBS v1.5 is released. In the meantime, you'll have to stick to using your 'fast' SD cards on your Pi1, or buy some 'slower' SD cards for use with your Pi2.

Kev-UK, 2015-07-07:
I have the same issue with a 32GB PNY Class 10 MicroSDHC 50MB/s, I get to the same error as crispcbk, used the SDFormatter tool v4.0 and performed a "Full (overwrite)" format. Brand new card straight out of the box. Perplexed at the moment.

qbasicer, 2015-07-14:
I'm having the same issue on a class 2 SanDisk 16GB card that's a number of years old now (2010?)

3wulution, 2015-07-18:
I´m having the same problem with a Transcend 32GB class 10 card. please solve!

frankduiker, 2015-07-26:
I received my brand new RP2 + power supply + Transcend 16 GB class 10 card yesterday and have the same problem as crispcbk : Setup process stops with the error message "mkfs.fat: warning - lowercase labels might not work properly with DOS or Windows".

I solved it by creating 2 partitions and writing the NOOBS distribution to the first partition. I used MiniTools Partition Wizard on Windows 7 to create and format the SD card.

screamzz, 2015-07-29:
Hello, 
have the same problem with the latest noobs with Verbatim 44083 like crispcbk. It was marked as OK on the wiki. I switched to another 32 Go San Disk ultra model and it seems ok for now .... Damn it :(

Ruffio, 2015-07-29:
What is the nature/root cause of this bug? It seems random for which SD cards that is affected by this.

lurch, 2015-07-29:
> What is the nature/root cause of this bug?

Please see the already-mentioned https://github.com/raspberrypi/firmware/issues/397

byctech, 2015-08-01:
Until v1.5, you need get a bunch of SD cards from different companies, try them out until you find one that works. For me, Patriot 8GB.

janww, 2015-08-12:
@frankduiker How excatly did you format the partitions? I have the exact same card and the same problem!

frankduiker, 2015-08-13:
@janww I created two partitions. The first partition is a primary partition, FAT32 of less then 8Gb (7.2 or something like that). I installed the NOOBS image on this partition.

noodle-, 2015-09-13:
For anyone still having the same issue, don't use the NOOBS installer.
I'm guessing NOOBS has problems with SD cards that are > 8GB. Just a guess though, based on @frankduiker 's solution.

I used SD Formatter 4.0 (with resize option ON, and no name for the SD card set), copied NOOBS v1.4.1 to the SD card, followed the on screen instructions to install Raspian three times, to no avail.
I got error creating file system errors every time.

I then downloaded the Raspian image, flashed this to the SD card and it worked right away.

So just download the latest Raspian image from [here](https://www.raspberrypi.org/downloads/raspbian/) and use the normal [installation guide](https://www.raspberrypi.org/documentation/installation/installing-images/windows.md)(for windows)

dustrider, 2015-10-04:
I had the same issue, as per the pic posted by crispcbk.
The pic is indicative of a FAT problem with the volume name. Fixed this by:
reformatting the card with the SD association formatting tool, 
Making sure to name the volume with all caps e.g. "PI"
quick format & size adjustment on

after this the normal install process worked fine.

CyborgHead, 2015-11-07:
I am having the same problem as in the screenshot by @crispcbk on a brand new Raspberry Pi 2 Model B.

Formatted a brand-new "microSDHC Transcend 32 GB Class 10 UHS-I 300x" using SDFormatter - with Format Adjustment = ON - on Windows 10 (x64), and then extracted NOOBS 1.4.2 (Offline Install) into the formatted memory card. Also, named the volume NOOBS - all upper-case - as @dustrider recommended, but no progress.