raspberrypi/noobs #231: option to override hardcoded DEFAULT_REPO_SERVER

beta-tester, 2015-02-10:
it would be nice, to be able to override the hardcoded
`#define DEFAULT_REPO_SERVER "http://downloads.raspberrypi.org/os_list_v2.json"`
for special purpose.
for example by adding an option in the config.txt or cmdline.txt file or any other file for noobs.

one of the special purpose is to download all the OS installation files once to a local server and add new or modify existing images to share/distribute them over the local network.
so all local RPi's can be installed quick and easily with the modified images in an education environment.

or sometimes the schools network firewall will not let the RPi to "go outside"...
so a local solution would be nice...
i know, preparing a "heavy loaded" NOOBS version SD card with all needed images on it would be a solution, but it is waste of SD card space and installation preparation time for the teachers...
a quick installation of a NOOBS lite and installation via local network would be possible for the childs in-time of a lession easily

procount, 2015-02-10:
I can do that for you. In fact I've already done it in my version, but it's based on 1.3.10, not the latest 1.3.12 for the Pi2. If that's suitable for you, I'll extract it from my extra mods and post a version up for you. Please allow a few days..... Would you prefer the updated source, or a pre-built version of noobs-lite?

beta-tester, 2015-02-10:
thank you...
if you have added that feature and done changes on the source-code already, then it is possibly a good idea, to make a kind of commit / pull request here, so your changes can go to the NOOBS master trunk...
so every body will benifit form your work... not only me.

procount, 2015-02-10:
Absolutely. But that may take a while because I think Lurch is a bit busy at the moment and it might have to align with RPF releases? In any case I will push it to my github fork, so anyone can build it from there and maybe Lurch can pull the changes through too. I'll post here when it's done. Just let me know if you need it prebuilt.

lurch, 2015-02-10:
I've also had this feature privately requested too... but in that case they wanted to be able to add their own server 'on top' of `DEFAULT_REPO_SERVER`, rather than replace it entirely. Any chance you could incorporate that option into your version too @procount, while you're at it? ;-)

procount, 2015-02-10:
I foresee a slippery slope ahead..... :-)

procount, 2015-02-19:
Please use Pull Request in #234 instead of #232. The latter could be deleted.

lurch, 2015-02-19:
@procount You can do that yourself - simply click on the "Close pull request" button on #232  :-)

beta-tester, 2015-02-20:
@procount and @lurch thank you alot...
I'm very curious to see how it will look like finally

lurch, 2015-02-21:
@procount maybe you could add a (very) brief bit of 'user documentation' to #234 so that e.g. @beta-tester or @digitalLumberjack could give some feedback on the new options you've provided?

procount, 2015-02-21:
Yep. No probs - in Readme.md?

lurch, 2015-02-21:
I was thinking just as comments in the 'Conversation' on #234 - probably 'more visible' to people there, than burying it in a modification to README.md ;-)
(but adding documentation to `README.md` at the same time wouldn't go amiss!)

tuupola, 2016-10-08:
Any news on this? I am using NOOBS for user friendly reinstall of application we are providing to our customers. Having option to override `DEFAULT_REPO_SERVER` without recompiling NOOBS would be excellent addition.

procount, 2016-10-08:
It's already in PINN - NOOBS with the extras... https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=142574

PINN is totally compatible with NOOBS. You can define your own repos in addition to the Rpf server to serve additional images, or just specify your own lical lan server.

Follow the link to download pinn-lite.zip from sourceforge and treat just like noobs-lite.

Pinn can also install images from a USB stick or HDD.

tuupola, 2016-10-08:
Awesome. Will check it out.

lurch, 2016-10-31:
Looks like this has been fixed by https://github.com/raspberrypi/noobs/commit/0feffba26cabf63e46577b1e6faf927c0644b097 ?

closed 2017-11-03 by XECDesign