raspberrypi/noobs #235: Security flaw: HTTPS not supported on download site

lukestanley, 2015-02-20:
To reproduce go to:
https://www.raspberrypi.org/downloads
You should see that it doesn't connect.

People on shared wifi often can't verify a download over HTTP and could be easily tricked into downloading malware to their Pi.

lurch, 2015-02-20:
Well, 'luckily' NOOBS doesn't yet support WiFi ;-)  #15 

Even if it did, I'm not sure how much of a concern this actually is, as even Raspbian (which *does* support wifi) only has HTTP links in its `sources.list` files rather than HTTPS.

So thanks for the report, but closing as invalid.

closed 2015-02-20 by lurch

lukestanley, 2015-02-20:
Hi lurch, I believe what I said has been misinterpreted and incorrectly closed. The problem is that the place where the image download links are doesn't serve HTTPS. Anyhow, source.list is different, as I believe those files are verified by other means.

lurch, 2015-02-20:
Ahhh, well the image download links are outside the scope of NOOBS itself, so this still doesn't count as a NOOBS bug. But I'll pass your concern along to the relevant people...

lukestanley, 2015-02-20:
Great. I did look at the contact page but didn't see a security contact or anything obviously similar.

lurch, 2015-04-10:
You'll be pleased to hear that the downloads page is now served over an HTTPS (SSL) link https://www.raspberrypi.org/downloads/ and therefore it's now possible to get the NOOBS SHA1 checksums without worrying that they've been interfered with, and thus you can verify that you've got a legitimate NOOBS download :-)

bennuttall, 2015-04-10:
The downloads page is HTTPS but the image downloads themselves are not, as we only have a certificate for the `www` subdomain.

lurch, 2015-04-10:
@bennuttall That's fine, I believe the checksums themselves being served over HTTPS (which means you can detect any corruption in the files downloaded over HTTP) is all that's needed.

bennuttall, 2015-11-22:
Re: above reference in #290 

We now have a wildcard SSL certificate covering the whole raspberrypi,org subdomain namespace.