raspberrypi/noobs #236: Editing the recovery.cmdline

johnkors, 2015-02-21:
Hi, I'm a NOOB (in several meanings). Having difficulting editing the recovery.cmdline, as my text editors says it's read-only. I tried modifying them with chmod, but still no luck.

I'm trying to use the "add forcetrigger to the arguments" trick to boot into NOOBS, as the hold-shift-down trigger didn't work for me. 

Any help?

Running latest NOOBS (updated today).

johnkors, 2015-02-21:
Nevermind, holding down the shift key only at the "grey screen" gave me the NOOBS UI.

closed 2015-02-21 by johnkors

lurch, 2015-02-21:
It sounds like possibly your SD card is being mounted in read-only mode? (from the mention of `chmod`, I'm assuming you're using Linux?)

At the command-line what does the output of `mount` report? You should be able to re-mount the NOOBS recovery partition as read-write with `sudo mount -o remount,rw {mountpoint}` where {mountpoint} might be something like `/media/RECOVERY`.

lurch, 2015-02-21:
Aha! Yeah, sometimes 'tapping' the shift key is more reliable than 'holding down' the SHIFT key ;-)

NicoHood, 2015-02-21:
And why? If there is no way to fix it, why not add a note like "press the shift key or tab it several times if its not working" ... with some better words of course.

lurch, 2015-02-21:
You mean like https://github.com/raspberrypi/noobs#what-to-do-if-your-shift-keypress-isnt-detected ?

johnkors, 2015-02-21:
Yep, I got the "forcetrigger" from that part of the docs. I missed the bit with only holding the shift key at the grey screen on boot. My bad.

NicoHood, 2015-02-21:
Yeah. But not in the docs, on the raspi screen. It might sound lazy, but hey, if its a bug and this might confuse noobs who dont read the docs. Or you could try to fix it, if possible.

johnkors, 2015-02-21:
The grey screen actually says this. "Hold the shift key down". I believe I was holding the shift key before the grey screen appeared, and that did not trigger NOOBS.

lurch, 2015-02-21:
AFAIK it's entirely keyboard-dependent whether holding down the SHIFT key from before power on, or only after the grey splash-screen appears, gets recognised by the HID driver in the Linux kernel. I've got USB keyboards that behave in both ways, so there's nothing that can be "fixed".

Do you have any suggestions for a better message to display on the splash-screen? If you can come up with something better than "Hammer the shift key like a maniac" ;-) then feel free to suggest it in a new issue.

NicoHood, 2015-02-21:
That is great humor!

+1 for "Hammer the shift key like a maniac"