raspberrypi/noobs #237: Add better boot shift key message

NicoHood, 2015-02-21:
Referring to issue #236 I'd like to see an additional message at bootup that tells the user that he should try to press the shift key several times rather then only holding it down, since that might not work. Lurch's idea is great I think, just add this as 2nd line (short, simply and funny):

"Or hammer the shift key like a maniac."

lurch, 2015-02-21:
Unfortunately, I don't think that'd be very appropriate.

(Feel free to re-open this bug if you come up with a serious suggestion)

NicoHood, 2015-02-21:
FYI even If I want to, i cannot reopen the issue once you've closed it.

You could change the text to:
"Trigger/press the shift button to enter <settings or whatever its named>"

You just need another word for "hold". Its like entering a bios on pc. Maybe simply "press" would also be enough. But holding is wrong in this case.

lurch, 2015-02-21:
Re-opened :)

johnkors, 2015-02-22:
I think adding a time aspect into that message, like "and do it NOW" would suffice. No need to involve MC Hammer ;)

NicoHood, 2015-02-22:
"For recovery mode, press Shift now."
Yeah that sounds good.

NicoHood, 2016-02-06:
Does anyone know where this text is set? I currently try to fix it, but I cannot find it. I tried so many greps, but it seems to be missing?

dubiouscript, 2019-03-16:
> Does anyone know where this text is set? I currently try to fix it, but I cannot find it. I tried so many greps, but it seems to be missing?

try strings ..? 
https://en.wikipedia.org/wiki/Strings_(Unix)