raspberrypi/noobs #238: Remove a single OS without wiping whole card

ghub25, 2015-02-22:
It would be great if there was a way to remove one of the OS that noobs has installed, without affecting the others.

For instance, say I am new to Raspberry Pi and install Raspbian, Raspbmc, and RetroPie. As I become more proficient I realize, I can just compile RetroPie and Kodi into Raspbian, and thus have any background services I might be using my pi for (like bittorrent sync serve) remain operational and uninterrupted while I watch a movie or play a game. That's great, but now a lot of space on my card is being used up by these 2 distros I no longer need, and I have to wait for the annoying time to go by for the raspberry pi to continue booting into Raspbian. Of course I could just do a fresh reinstall and only add Raspbian, but all those hours and hours of configuration and 10s of hours of compiling from source will have to be redone! It would be awesome if I could just remove the other OS I no longer use :).

lurch, 2015-02-22:
Nice idea, but it would require NOOBS to be able to shuffle / move / resize the partitions, which is well beyond the scope of what NOOBS does currently.

For example if you had
/dev/mmcblk0p5  Raspbian boot
/dev/mmcblk0p6  Raspbian root
/dev/mmcblk0p7  Retropie boot
/dev/mmcblk0p8  Retropie root
then what would NOOBS be expected to do if you deleted Raspbian?

You can do what you're asking for manuallly using something like GParted on a regular Linux PC (and then edit the files on the /dev/mmcblk0p3 SETTINGS partition on the SD card to reflect the new partition numbers and available OSes), or you should be able to make use of https://github.com/raspberrypi/noobs#how-to-create-a-custom-os-version to backup your current OS and then 're-install' it as the only OS.

closed 2015-02-22 by lurch

tommysuryajaya, 2018-06-03:
Hi
Just delete your OS that you want when you select OS while booting ur raspberry pi
you can delete it from setting 

i hv try to delete ubuntu because its using a lot of memory, and success

moebaa19, 2019-07-22:
> while booting ur raspberry pi
you can delete it from setting

@tommysuryajaya sorry but I really dont know how to do it... thx for answering

lurch, 2019-07-22:
I believe https://github.com/procount/pinn is now able to do something similar to what's being asked for here. ping @procount to provide more details :wink:

procount, 2019-07-22:
PINN will make it a bit easier, but it's still not a very well supported feature, because it means moving partitions around and neither PINN nor NOOBS is a substitute for gparted.

Using PINN, I would suggest backing up the OSes that you want to keep to a USB stick, then install just the backups you want, leaving out the ones you don't. It's easy to do, it will just take a while to do all the copying.

However, I suppose you are currently using NOOBS. Whilst PINN is compatible with NOOBS, some of its newest features rely on PINN having installed the OSes, so I'm not 100% sure if this will work. 

In any case no matter how you intend to proceed, Step #1 is always to **MAKE A COMPLETE BACKUP OF YOUR EXISTING SD CARD** (e.g. using Raspbian's SD card copier) and **work on your backup**, so it doesn't matter if things go wrong :wink: