raspberrypi/noobs #241: Raspberry Pi 2 - Arch

rjgould, 2015-02-26:
Arch appears to have support for Pi 2 under a separate branch, can this be added to noobs?

http://archlinuxarm.org/platforms/armv7/broadcom/raspberry-pi-2

lurch, 2015-02-26:
I'd be happy to, if @kmihelich (or another one of the ArchLinuxARM developers) wants to provide me with a NOOBS-installable version?

Zeitwaechter, 2015-03-09:
+1 for workflow simplicity to get Arch to run on a Pi2.
Personally most important distribution. Won't use anything else besides Arch.

BTW: Thanks for all your work on NOOBS, @lurch !

pickfire, 2015-03-30:
I hope that it has arch for Pi 2 too. Anything I can help?

Zeitwaechter, 2015-08-06:
*push*

Any news? Does it work OOTB now?

beta-tester, 2015-09-29:
bump...

i really really would like to see a more recent ARCH version for RPi1 and RPi2.
i think there were some communication misunderstands between the NOOBS Team and the ARCH Team in the past...
but i hope both parties to come back to each other and start a new try to help end-users.

thank you very much!

beta-tester, 2015-09-29:
as suggestion, to "transform" an ArchLinuxARM-rpi-latest.tar.gz image
to a NOOBS compatible boot/root.tar.xz image:
```sh
wget http://os.archlinuxarm.org/os/ArchLinuxARM-rpi-latest.tar.gz
mkdir -p /tmp/boot
mkdir -p /tmp/root
tar -xvpaf ArchLinuxARM-rpi-latest.tar.gz -C /tmp/root
cp -r /tmp/root/boot/* /tmp/boot/.
rm -rf /tmp/root/boot/*
cd /tmp/boot/
tar -cvpaf ../boot_pi1.tar.xz .
cd /tmp/boot/
tar -cvpaf ../root_pi1.tar.xz .
rm -rf /tmp/boot/
rm -rf /tmp/root/
```

```sh
wget http://os.archlinuxarm.org/os/ArchLinuxARM-rpi-2-latest.tar.gz
mkdir -p /tmp/boot
mkdir -p /tmp/root
tar -xvpaf ArchLinuxARM-rpi-latest.tar.gz -C /tmp/root
cp -r /tmp/root/boot/* /tmp/boot/.
rm -rf /tmp/root/boot/*
cd /tmp/boot/
tar -cvpaf ../boot_pi2.tar.xz .
cd /tmp/boot/
tar -cvpaf ../root_pi2.tar.xz .
rm -rf /tmp/boot/
rm -rf /tmp/root/
```
and adjusting some other NOOBS files:
http://downloads.raspberrypi.org/os_list.json
http://downloads.raspberrypi.org/os_list_v2.json
```json
        {
            "description": "An Arch Linux port for ARM devices",
            "feature_level": 35120124,
            "icon": "http://downloads.raspberrypi.org/arch/Arch.png",
            "marketing_info": "http://downloads.raspberrypi.org/arch/marketing.tar",
            "nominal_size": 1600,
            "os_info": "http://downloads.raspberrypi.org/arch/os_pi1.json",
            "os_name": "Arch",
            "partition_setup": "http://downloads.raspberrypi.org/arch/partition_setup.sh",
            "partitions_info": "http://downloads.raspberrypi.org/arch/partitions.json",
            "release_date": "2015-09-03",
            "supported_hex_revisions": "2,3,4,5,6,7,8,9,d,e,f,10,11,12,14,19",
            "tarballs": [
                "http://downloads.raspberrypi.org/arch/boot_pi1.tar.xz",
                "http://downloads.raspberrypi.org/arch/root_pi1.tar.xz"
            ]
        },
        {
            "description": "An Arch Linux port for ARM devices",
            "feature_level": 0,
            "icon": "http://downloads.raspberrypi.org/arch/Arch.png",
            "marketing_info": "http://downloads.raspberrypi.org/arch/marketing.tar",
            "nominal_size": 1600,
            "os_info": "http://downloads.raspberrypi.org/arch/os_pi2.json",
            "os_name": "Arch",
            "partition_setup": "http://downloads.raspberrypi.org/arch/partition_setup.sh",
            "partitions_info": "http://downloads.raspberrypi.org/arch/partitions.json",
            "release_date": "2015-09-03",
            "supported_hex_revisions": "1040,1041",
            "tarballs": [
                "http://downloads.raspberrypi.org/arch/boot_pi2.tar.xz",
                "http://downloads.raspberrypi.org/arch/root_pi2.tar.xz"
            ]
        },
```

http://downloads.raspberrypi.org/arch/os_pi1.json
for RPi1 version only
```json
{
  "name": "Arch",
  "url": "http://archlinuxarm.org/platforms/armv6/raspberry-pi",
  "version": "20150903",
  "release_date": "2015-09-03",
  "kernel": "4.x",
  "description": "An Arch Linux port for ARM devices",
  "username": "root",
  "password": "root",
  "feature_level": 35120124
}
```
http://downloads.raspberrypi.org/arch/os_pi2.json
for RPi2 version only
```json
{
  "name": "Arch",
  "url": "http://archlinuxarm.org/platforms/armv7/broadcom/raspberry-pi-2",
  "version": "20150903",
  "release_date": "2015-09-03",
  "kernel": "4.x",
  "description": "An Arch Linux port for ARM devices",
  "username": "root",
  "password": "root",
  "feature_level": 0
}
```

http://downloads.raspberrypi.org/arch/partitions.json
can be shared by RPi1 and RPi2 version
```json
{
  "partitions": [
    {
      "label": "boot",
      "filesystem_type": "FAT",
      "partition_size_nominal": 100,
      "want_maximised": false,
      "uncompressed_tarball_size": 20,
      "mkfs_options": "-F 32"
    },
    {
      "label": "root",
      "filesystem_type": "ext4",
      "partition_size_nominal": 1500,
      "want_maximised": true,
      "mkfs_options": "-c",
      "uncompressed_tarball_size": 600
    }
  ]
}
```

all other files can be taken from the old http://downloads.raspberrypi.org/arch/ folder.
http://downloads.raspberrypi.org/arch/Arch.png
http://downloads.raspberrypi.org/arch/marketing.tar
http://downloads.raspberrypi.org/arch/partition_setup.sh

NicoHood, 2016-02-07:
We got a working version here:
https://github.com/raspberrypi/noobs/issues/288

Ubuntu Mate also works:
https://github.com/raspberrypi/noobs/issues/307

XECDesign, 2017-11-03:
Arch does not support NOOBS.

closed 2017-11-03 by XECDesign