raspberrypi/noobs #242: No default gateway in route

hakito, 2015-03-01:
Noobs: 1.4

Just downloaded and booted my raspberry pi with noobs.

I get an error in the gui that no distributions could be loaded. When I log into the recovery console and type route. I can see that my pi got the correct IP but not the correct default gateway. When I setup the correct gateway and I switch to the graphical console, I can now use the integrated browser, but can't get the installer to re-download the distribution list.

All other computers on my network Windows/Linux/Mac/iOS have no problem getting the gateway and dns from my dhcp server.

lurch, 2015-03-02:
Hmmm, first time I've heard of any problem like this! :-/
What router / dhcp server are you using on your network? I assume both NOOBS v1.4.0 and NOOBS Lite v1.4 have the same behaviour?

As a test, could you try [NOOBS Lite v1.3.10](http://downloads.raspberrypi.org/NOOBS_lite/images/NOOBS_lite-2014-09-12/) as that was the last version using the old 3.6.y kernel, whereas all versions since then are using a newer 3.18.y kernel and newer buildroot infrastructure.

lurch, 2015-04-07:
...any more info @hakito ? Otherwise I may have to close this issue due to lack of activity.

lurch, 2015-05-04:
Any updates, @hakito ?

hakito, 2015-05-04:
Hi, sorry for the long inactivity. I am using ubuntu dhcp3-server 4.1.ESV-R4-0ubuntu5.9

I didn't try the lite version. What I can tell is, that i also had this problem when using some other linux distribution package directly.

I should probably mention that I'm not using x.x.x.1 but x.x.x.81 as gateway oon my network. The netmask is 255.255.255.0.

lurch, 2015-05-05:
When you say that "i also had this problem when using some other linux distribution package directly" do you mean that you tried other DHCP server packages, and they also caused the same problem with NOOBS; or do you mean that other standalone Linux distros also had a problem getting the correct default gateway address from this particular server?
I'll have to find time to set up a standalone test network to try to reproduce this, but as this is low priority (since you're the only person to report this) I dunno when that will be. In the meantime, if you could try out both NOOBS Lite v1.4 and NOOBS Lite v1.3.10 (using the download link provided above) to see if they have the same problem, that would be really useful.

hakito, 2015-05-16:
I mean other standalone distros also had the problem. I did not try other DHCP servers. I tried now with noobs 1.3.10 lite and noobs 1.4. lite. Both had network access but did not get the gateway. I tried to capture the traffic for the DHCP communication, but i could only capture the broadcast from the pi.

hakito, 2015-05-16:
I tried now plugging a standard zyxel router between my internal network and the raspberry pi. The zyxel router acts as it's own DHCP server on another subnet. (192.1.1.*)  So either the rasberry has problems with the Ubuntu DHCP or with the network 10.54.217.*

AS I said before I have a lot of other computers, printers cubox-i wlan clients etc. that retrieve their IP addresses, Gateway and nameserver settings without any problems. So it's really weird that just the raspberry seems to be affected.

celevra, 2016-02-10:
i'm using windows server 2012 R2 DHCP Server and have the same problem.
I have to switch to console, kill the recovery. Add the route and start recover -runinstaller -qws

redkite1, 2016-09-05:
Hi, I exactly got the same issue here, no default gateway received from my DHCP. It's working fine for my other distros.

Problematic setup:
- Raspbian Jessie 4.4
- Router: OpenWRT
- Network: 10.161.0.x/24

I'll give you an other comment if I have a chance to try with an other router.

maxnet, 2016-09-05:
Your server doesn't send anything else either that the client may favor above the standard router option, e.g. a list of static routes?
If you do ``dhcpcd -T`` there is no new_routers variable?

redkite1, 2016-09-08:
You are right, I also send one static route to another sub-network (10.161.1.x/24 gw 10.161.0.253).

Here is the result of route:
```
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
10.161.0.0      *               255.255.255.0   U     202    0        0 eth0
10.161.1.0      10.161.0.253    255.255.255.0   UG    0      0        0 eth0
10.161.1.0      10.161.0.253    255.255.255.0   UG    202    0        0 eth0
```

Here is the result of dhcpcd -T eth0:
```
$ sudo dhcpcd -T eth0
dhcpcd[1144]: version 6.7.1 starting
dhcpcd[1144]: eth0: IPv6 kernel autoconf disabled
dhcpcd[1144]: DUID 00:01:00:01:1e:da:f0:0a:b8:27:eb:0a:3e:05
dhcpcd[1144]: eth0: IAID eb:3d:1e:55
dhcpcd[1144]: eth0: soliciting an IPv6 router
dhcpcd[1144]: eth0: rebinding lease of 10.161.0.16
interface='eth0'
pid='1144'
reason='TEST'
skip_hooks='lookup-hostname'
ifcarrier='up'
ifflags='4163'
ifmtu='1500'
ifwireless='0'
new_broadcast_address='10.161.0.255'
new_classless_static_routes='10.161.1.0/24 10.161.0.253'
new_dhcp_lease_time='43200'
new_dhcp_message_type='5'
new_dhcp_rebinding_time='37800'
new_dhcp_renewal_time='21600'
new_dhcp_server_identifier='10.161.0.254'
new_domain_name='lan'
new_domain_name_servers='10.161.0.254'
new_host_name='retropie'
new_ip_address='10.161.0.16'
new_network_number='10.161.0.0'
new_routers='10.161.0.254'
new_subnet_cidr='24'
new_subnet_mask='255.255.255.0'
old_broadcast_address='10.161.0.255'
old_classless_static_routes='10.161.1.0/24 10.161.0.253'
old_dhcp_lease_time='43200'
old_dhcp_message_type='5'
old_dhcp_rebinding_time='37800'
old_dhcp_renewal_time='21600'
old_dhcp_server_identifier='10.161.0.254'
old_domain_name='lan'
old_domain_name_servers='10.161.0.254'
old_host_name='retropie'
old_ip_address='10.161.0.16'
old_network_number='10.161.0.0'
old_routers='10.161.0.254'
old_subnet_cidr='24'
old_subnet_mask='255.255.255.0'
dhcpcd[1144]: exited
```

maxnet, 2016-09-08:
>You are right, I also send one static route to another sub-network (10.161.1.x/24 gw 10.161.0.253).

Let's see what the specification says about what DHCP clients capable of understanding such advanced network configurations that include a static route list are supposed to do in such situation.

https://tools.ietf.org/html/rfc3442

```
   If the DHCP server returns both a Classless Static Routes option and
   a Router option, the DHCP client MUST ignore the Router option.
```

So looks like dhcpcd is acting correctly.
If you do want 10.161.0.254 to play default gateway and handle other routes, I think adding 
 "0.0.0.0/0 10.161.0.254" to the "static" route list may do what you want, although I have not tested it.

zakalawe, 2017-05-26:
Encountering a similar issue with, all other Windows/Mac/Ubuntu clients are fine, but Jessie won't add DNS or route. 'dhcpcd -T' shows a 'new_routers' and new_domain_servers values which look good.

When I run 'systemctl start dhcpcd.service' and then 'status' to see the info, I notice in the output:

'eth0: soliciting an IPv6 router'
'eth0: rebinding lease of 10.0.1.25'
'eth0: changing route to 10.0.1.0/24'
'eth0: removing route to 10.0.1.0/24' 

This makes me wonder if the IPv6 solicitation is removing the IP4 routes? By router / DHCP server is an Apple AirPort Extreme which can do IPv6 in theory, but my ISP doesn't.

maxnet, 2017-05-26:
>This makes me wonder if the IPv6 solicitation is removing the IP4 routes? 

Doubt that, but check by putting ``ipv4only`` in dhcpcd.conf

It's a fresh Jessie installation?
You didn't install -directly or indirectly through dependencies- any other software packages that may act as DHCP client or network configuration management tool and conflict?

zakalawe, 2017-05-26:
It's stock Jessie, but the pointer to examine dhcpcd.conf helped: there were some weird entries:

`static routers=
static domain_name_servers=
static domain_search=
`

Removing these makes everything work as normal. I don't believe I added these entries, I did try to fix this issue using the GUI tool (from the menubar). The GUI appeared to have no effect, which files is it supposed to modify?

maxnet, 2017-05-26:
>The GUI appeared to have no effect, which files is it supposed to modify?

That one.

zakalawe, 2017-05-28:
Okay, my suspicion is that the GUI 'clear' button was removing the values but not the lines in the file. But since I fixed the file manually, I did an update and the current version of the GUI now exhibits the correct behaviour: add manual entries, they are set in dhcpcd.conf. Then press 'clear' in the GUI and lines are removed.

So, no more bug for me anyway.

closed 2018-03-08 by XECDesign