raspberrypi/noobs #244: Caanot rebuil NOOBS v1.4

adontz, 2015-03-02:
Tried with 1.4 release and latest from git and got one and the same error trying to compile qjson. The following is corresponding build log fragment. I'm building on latest Arch x64.

    -- Build files have been written to: /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02
    >>> qjson ba273682a9d33a7b3090e74f4742b5f3bf6c9b02 Building
    PATH="/data/noobs-1.4/buildroot/output/host/bin:/data/noobs-1.4/buildroot/output/host/sbin:/data/noobs-1.4/buildroot/output/host/usr/bin:/data/noobs-1.4/buildroot/output/host/usr/sbin:/usr/local/sbin:/usr/lo                              cal/bin:/usr/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl"  /usr/bin/make -j3  -C /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/
    make[1]: Entering directory '/data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02'
    make[2]: Entering directory '/data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02'
    make[3]: Entering directory '/data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02'
    Dependee "/data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/CMakeFiles/qjson_automoc.dir/DependInfo.cmake" is newer than depender "/data/noobs-1.4/buildroot/output/bui                              ld/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/CMakeFiles/qjson_automoc.dir/depend.internal".
    Dependee "/data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/data/noobs-1.4/buildroot/output/build/                              qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/CMakeFiles/qjson_automoc.dir/depend.internal".
    Scanning dependencies of target qjson_automoc
    make[3]: Leaving directory '/data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02'
    make[3]: Entering directory '/data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02'
    [ 11%] Automatic moc for target qjson
    AUTOGEN: Checking /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/parser.cpp
    AUTOGEN: Checking /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/qobjecthelper.cpp
    AUTOGEN: Checking /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/json_scanner.cpp
    AUTOGEN: Checking /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/json_parser.cc
    AUTOGEN: Checking /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/parserrunnable.cpp
    AUTOGEN: Checking /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/serializer.cpp
    AUTOGEN: Checking /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/serializerrunnable.cpp
    AUTOGEN: Checking /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/json_parser.hh
    AUTOGEN: Checking /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/json_scanner.h
    AUTOGEN: Checking /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/parser.h
    AUTOGEN: Checking /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/parser_p.h
    AUTOGEN: Checking /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/parserrunnable.h
    AUTOGEN: Checking /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/qjson_export.h
    AUTOGEN: Checking /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/qobjecthelper.h
    AUTOGEN: Checking /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/serializer.h
    AUTOGEN: Checking /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/serializerrunnable.h
    Generating moc_parserrunnable.cpp
    /usr/lib/qt/bin/moc -I/data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src -I/usr/include/qt -I/usr/include/qt/QtCore -I/usr/lib/qt/mkspecs/linux-g++ -I/data/noobs-1.4/bu                              ildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/. -DQT_CORE_LIB -DQT_NO_DEBUG -DQT_NO_DEBUG_OUTPUT -o /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c                              9b02/src/moc_parserrunnable.cpp /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/parserrunnable.h
    Generating moc_qobjecthelper.cpp
    /usr/lib/qt/bin/moc -I/data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src -I/usr/include/qt -I/usr/include/qt/QtCore -I/usr/lib/qt/mkspecs/linux-g++ -I/data/noobs-1.4/bu                              ildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/. -DQT_CORE_LIB -DQT_NO_DEBUG -DQT_NO_DEBUG_OUTPUT -o /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c                              9b02/src/moc_qobjecthelper.cpp /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/qobjecthelper.h
    /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/qobjecthelper.h:0: Note: No relevant classes found. No output generated.
    Generating moc_serializerrunnable.cpp
    /usr/lib/qt/bin/moc -I/data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src -I/usr/include/qt -I/usr/include/qt/QtCore -I/usr/lib/qt/mkspecs/linux-g++ -I/data/noobs-1.4/bu                              ildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/. -DQT_CORE_LIB -DQT_NO_DEBUG -DQT_NO_DEBUG_OUTPUT -o /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c                              9b02/src/moc_serializerrunnable.cpp /data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/serializerrunnable.h
    make[3]: Leaving directory '/data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02'
    [ 11%] Built target qjson_automoc
    make[3]: Entering directory '/data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02'
    Dependee "/data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/CMakeFiles/qjson.dir/DependInfo.cmake" is newer than depender "/data/noobs-1.4/buildroot/output/build/qjson                              -ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/CMakeFiles/qjson.dir/depend.internal".
    Dependee "/data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/data/noobs-1.4/buildroot/output/build/                              qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/src/CMakeFiles/qjson.dir/depend.internal".
    Scanning dependencies of target qjson
    make[3]: Leaving directory '/data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02'
    make[3]: Entering directory '/data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02'
    [ 22%] [ 33%] [ 44%] Building CXX object src/CMakeFiles/qjson.dir/qobjecthelper.cpp.o
    Building CXX object src/CMakeFiles/qjson.dir/parser.cpp.o
    Building CXX object src/CMakeFiles/qjson.dir/json_scanner.cpp.o
    cc1plus: error: include location "/usr/include/qt" is unsafe for cross-compilation [-Werror=poison-system-directories]
    cc1plus: error: include location "/usr/include/qt/QtCore" is unsafe for cross-compilation [-Werror=poison-system-directories]
    cc1plus: some warnings being treated as errors
    cc1plus: error: include location "/usr/include/qt" is unsafe for cross-compilation [-Werror=poison-system-directories]
    cc1plus: error: include location "/usr/include/qt/QtCore" is unsafe for cross-compilation [-Werror=poison-system-directories]
    cc1plus: some warnings being treated as errors
[... cut ...]
    make[2]: Leaving directory '/data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02'
    Makefile:117: recipe for target 'all' failed
    make[1]: *** [all] Error 2
    make[1]: Leaving directory '/data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02'
    package/pkg-generic.mk:182: recipe for target '/data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/.stamp_built' failed
    make: *** [/data/noobs-1.4/buildroot/output/build/qjson-ba273682a9d33a7b3090e74f4742b5f3bf6c9b02/.stamp_built] Error 2

lurch, 2015-03-04:
Hmmm, I tried reproducing this, but was unable to.
I've never used ArchLinux before, so following [these instructions](https://wiki.archlinux.org/index.php/Installation_guide) I installed ArchLinux x64 from `archlinux-2015.03.01-dual.iso` into a VirtualBox VM. After repeated trial'n'error, I found that these packages needed to be installed: git, make, gcc, flex, bison, unzip, rsync, patch, wget, cpio, python, bc, qt4, pkg-config
And `recovery/updateqm.sh` needed to be modified to call `lupdate-qt4` and `lrelease-qt4` (if I remember, I'll modify updateqm.sh to try both the 'normal' and '-qt4' versions of the commands).

But `./BUILDME.sh` then ran to completion, without any build errors. (repeated twice just to be sure)
Any idea what might make your setup different, and how I can reproduce your problem?

lurch, 2015-03-04:
Aha... looks like if `qt5-base` is *also* installed, then qjson is choosing the system-installed QT5 headers, rather than using the buildroot-provided QT4 headers :-/  (which it shouldn't be doing when cross-compiling)
I'll dig into this some more later... it might be a [Buildroot](http://buildroot.uclibc.org/) bug.
(I've just checked, and https://github.com/raspberrypi/noobs/tree/master/buildroot/package/qjson is currently identical to http://git.buildroot.net/buildroot/tree/package/qjson and bumping QJSON_VERSION to 039961b58558c82867f9676486e21a50e8d939bf also produces the same build-error).

adontz, 2015-03-05:
Yes, I have Qt5 installed.

procount, 2015-09-17:
try export QT_SELECT=4. see #158

commit f2b4fc9e (2017-11-03): Set QT_SELECT=4

Fixes #244

Fixes #158
closed 2017-11-03 by XECDesign