raspberrypi/noobs #245: NOOBS crashes on Raspberry Pi A+

IvanExpert, 2015-03-05:
When you start NOOBS on a Raspberry Pi A+, it immediately crashes after the multicolored square, and says:

sh: can't access tty: job control turned off
/ # 

I have tried NOOBS 1.4.0 both regular and Lite, as well as NOOBS Lite 1.3.12 and 1.3.10. Same story in all cases.

lurch, 2015-03-05:
Do you have anything connected to your GPIO pins? Sounds like pin 5 (i.e. GPIO 3) is being pulled low, and triggering NOOBS' ["safe mode"](https://github.com/raspberrypi/noobs#how-to-boot-into-safe-mode). You can prevent this by adding `disablesafemode` to the end of `recovery.cmdline` (keeping the whole file just on one line). I've just added a note about this to the README.

IvanExpert, 2015-03-05:
Definitely nothing attached to any GPIO pins, but `disablesafemode` did the trick. Maybe I have a strange  A+, but visually it appears completely like it did in the box, so it might be worth double-checking on your end. But as far as I go, I'm fine now. Thanks.

closed 2015-03-05 by IvanExpert

lurch, 2015-03-08:
I've just double-checked, and NOOBS v1.4.0 works fine on my A+. So it sounds like your GPIO pins must be faulty or damaged.