raspberrypi/noobs #246: No HDMI or composite

jsavage, 2015-03-06:
Have just received a new Raspi V2 and a fresh Noobs card from CPC. No video HDMO or composite. tried pressing 1,2,3,4.  Also Return followed by 1,2.
Inserting a known good SD with OpenElec booted correctly and displayed video as expected. Using a known good PSU.
Is this a an isolated case or a known issue?

NicoHood, 2015-03-06:
maybe you did not use fat32?

jsavage, 2015-03-06:
This was a pre-prepared card from CPC so I had no need to format it.  Before using it in the pi, I captured a backup of the contents of the card on my PC using sudo dd bs=4M if=/dev/sdc of=newnoobscard.img
Does anyone know if Noobs has an embedded version number or if it writes any diagnostics to the card that might help?

NicoHood, 2015-03-06:
then try the newest noobs version. The older versions only work for model B/B+ but not B2

lurch, 2015-03-06:
Please let me know what the `BUILD-DATA` file says. What's the exact size (in bytes) of your `newnoobscard.img` backup? Did you get *any* output at all (e.g. rainbow square at initial power-on) or just a black screen?

jsavage, 2015-03-07:
@lurch Black output apart from a couple of very brief occurrence of random lines in the centre of the screen on composite. hdmi completely black. 

-rw-r--r-- 1 root root 7948206080 Mar  6 10:25 newnoobscardbackup.img
 sha1sum d9c6f0d5159813ff2b01757df8016ace49f8d78c
~/raspi-backups$ sha1sum newnoobscard.img 
d9c6f0d5159813ff2b01757df8016ace49f8d78c  newnoobscardbackup.img

have looked but cannot find BUILD-DATA file - where should I be looking?

@NicoHood - I know I can download a new copy of the latest Noobs image - that does not address, identify or correct the problem in a way that might help others

James

lurch, 2015-03-07:
> have looked but cannot find BUILD-DATA file - where should I be looking?

It should be in the root directory of the first partition on the sdcard (which should be in fat32 format). What files can you see when you mount the first partition on the card? What does `fdisk -l /dev/sdX` report for the card? (replace sdX as appropriate)

Is http://cpc.farnell.com/raspberry-pi/rpi2-modb-8gb-noobs/sbc-rpi-2-model-b8gb-noobs/dp/SC13795 the bundle you bought?

jsavage, 2015-03-07:
Yes it was order code SC13795 
OK found BUILD-DATA 
/mnt$ ls -l
total 24512
-rwxr-xr-x 1 root root    17856 Feb 18 16:26 bootcode.bin
-rwxr-xr-x 1 root root      303 Feb 18 16:26 BUILD-DATA
drwxr-xr-x 3 root root     4096 Feb 17 12:45 defaults
-rwxr-xr-x 1 root root     2250 Feb 18 16:23 INSTRUCTIONS-README.txt
drwxr-xr-x 4 root root     4096 Feb 17 14:14 os
-rwxr-xr-x 1 root root  2148120 Feb 18 16:24 recovery7.img
-rwxr-xr-x 1 root root       40 Feb 18 16:35 recovery.cmdline
-rwxr-xr-x 1 root root   554520 Feb 18 16:26 recovery.elf
-rwxr-xr-x 1 root root        0 Feb 18 16:26 RECOVERY_FILES_DO_NOT_EDIT
-rwxr-xr-x 1 root root  2100464 Feb 18 16:26 recovery.img
-rwxr-xr-x 1 root root 20238331 Feb 18 16:26 recovery.rfs
-rwxr-xr-x 1 root root     9728 Feb 18 16:23 riscos-boot.bin

/mnt$ cat BUILD-DATA
Build-date: 2015-02-18
NOOBS Version: v1.4
NOOBS Git HEAD @ be1862829eabda6c88856348920d545827b1e82b
rpi-userland Git master @ c5341f09dc686c17966369485f7dd27f58dc081c
rpi-firmware Git master @ 8aca5762f984f6decbeda294cee8418966c3d8d3
rpi-linux Git rpi-3.18.y @ b630db9dcb5c73367050a672823047b0466b16b9

sudo fdisk -l /dev/sdc
Disk /dev/sdc: 7948 MB, 7948206080 bytes
4 heads, 32 sectors/track, 121280 cylinders, total 15523840 sectors
Units = sectors of 1 * 512 = 512 bytes

Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disk identifier: 0x1660e00a

   Device Boot      Start         End      Blocks   Id  System
/dev/sdc1            8192     1673828      832818+   c  W95 FAT32 (LBA)
/dev/sdc2         1679360    15333375     6827008    5  Extended
/dev/sdc3        15333376    15398911       32768   83  Linux
/dev/sdc5         1687552     1810431       61440    c  W95 FAT32 (LBA)
/dev/sdc6         1818624    15333375     6757376   83  Linux

lurch, 2015-03-07:
Hmmm, that all looks as expected, so I'm not sure why it's not working.
SC13795 is currently out of stock, so I've just ordered SC13797 (which looks like the identical card) to see if I get the same problem too.

lurch, 2015-03-11:
My [SC13797](http://cpc.farnell.com/sandisk/sdsdqu-008g-bmut/microsd-card-8gb-class-10-noobs/dp/SC13797) arrived this morning, and works totally fine in my Pi2.

One thing to be aware of (which is perhaps what's catching you out?) is that these cards are actually NOOBS-with-Raspbian-preinstalled, and are configured to output over HDMI by default. The fact that Raspbian comes pre-installed means that it automatically auto-boots into Raspbian, rather than simply displaying the NOOBS 'choose an OS to install' screen (as a fresh install of NOOBS would do).
So if you want to use one of these cards with composite video output you'll need to:

1. Connect power to the Pi and rapidly start tapping shift on your keyboard (to force the recovery menu to load)
2. Wait for the green LED to stop blinking (should only take a few seconds)
3. Press '3' or '4' on your keyboard to select PAL or NTSC output as appropriate
4. Use the NOOBS config editor to edit Raspbian's [config.txt](http://www.raspberrypi.org/documentation/configuration/config-txt.md) for composite video output (remembering to delete any HDMI options)

jsavage, 2015-03-11:
Andrew,
Many thanks for that. That is definitely not how I expected the Noobs card
to be delivered and is not how they are described online or on their
packaging  I cant attempt to reproduce this problem just now but it may
explain why I had issues because initially, I did try the pi out on a TV
with composite only.  I certainly was not expecting to have to press Shift.
However, when I had no joy I then moved it to another TV and connected it
with an HDMI lead and a composite lead -> scart at the same time before
applying power. During the boot sequence and well after booting should have
finished,  I selected hdmi and scart as source in turn. Both screens were
black. I can only surmise that something got corrupted and the software had
hung.

Thanks again for checking.
James

lurch, 2015-03-11:
> That is definitely not how I expected the Noobs card
to be delivered and is not how they are described online or on their
packaging

Apparently many new users were complaining that the initial setup process of installing Raspbian to a new NOOBS card was taking too long, and therefore dampening the out-of-the-box enthusiasm; so that's why the NOOBS cards sold by some distributors now ship with Raspbian pre-installed.

It's _possible_ that your Raspbian install has become corrupted due to an unclean shutdown, but there's no way for NOOBS itself to get corrupted as it boots read-only. So if you just hold down (or rapidly tap) shift after the next power-on (as described above), then you should find NOOBS still fully-functional allowing you to install Raspbian afresh.

I'll close this issue now, but please feel free to add followup comments if you're still having problems.

closed 2015-03-11 by lurch