raspberrypi/noobs #250: Is it possible to interact with the NOOBS install remotely (VNC, SSH or other)?

Basilmir, 2015-03-12:
I copied the NOOBS files to a FAT32 microSD Card. There is no monitor or keyboard installed on the Raspberry Pi all I have is power and a network cable.

I've put the NOOBS microSD Card in and powered it up. Presumably it now "displays" the NOOBS OS installer but there is no way to know.

Can I connect to it now via VNC (GUI preferably) or SSH and tell it to install something?

The scenario I'm looking for is similar to setting up a Mac Mini Server, which is headless by default.

procount, 2015-03-12:
VNC is already supported (see #116). SSH was supported, then taken out, but I have built an enhancement that re-includes it again, available on my github if you are able to rebuild noobs,

Basilmir, 2015-03-12:
I see I need to add "vncinstall" to recovery.cmdline. Is that before or after the line that is already there?

procount, 2015-03-12:
Insert it in the line somewhere. recovery.cmdline should just consist of 1 long line. 
The first time you run NOOBS it will stop in the installation screen, which will allow you plenty of time to connect via VNC. After you have installed an OS, it will only stop on the recovery screen for a short time before running the installed OS. I therefore find it convenient to add 'forcetrigger' as well so that it always stops in noobs and I can connect VNC easily.
Noobs will get an IP address via DHCP, so you need to know what this IP address is (usually from your router, or ping available addresses etc).

Basilmir, 2015-03-12:
I see. This would be great if it were the default scenario. You see, this way setting up a Raspberry becomes very similar to setting up a Wifi Router. I plug it in and expect it to "come online" and then connect to it to set it up. It would be great to have this all over Raspbian too, connecting to it remotely becoming the standard but that might be a "no go" with some people.

It is however a must for NOOBS, I think. It just works!

procount, 2015-03-12:
The VNC option in NOOBS is a simple implementation used as an <i>alternative</i> to using the HDMI or composite screen. It is not a full implementation that you can use <i>as well</i> as a connected monitor. So it cannot be the default as you would get no display on a connected monitor, which would be the majority of use cases. It is largely undocumented unless you search the forum and this github. Its main use was for taking screenshots. 

By default Raspbian has SSH enabled, so you can easily connect to that once you have it installed. Adding Tightvncserver to Raspbian is just an apt-get away.

Basilmir, 2015-03-12:
Hmm... 
So my first line says this "runinstaller quiet vt.cur_default=1 elevator=deadline vncinstall" without the quotes. Put the card in the Pi and plugged the power and network cable. I see some activity on the green LED. Now I'm waiting for it to request a DHCP ip...

procount, 2015-03-12:
Nope. And Noobs won't tell you what it is either.

There are several ways to find the IP address.
1. open the webpage on the router that has your DHCP server and display the DHCP client list. This varies depending on your router, but look for a hostname of RaspberryPi or similar.
2. Configure your DHCP server to provide a specific IP address for the RaspberryPi beforehand, 
3. Use a tool like ipscan to scan your network for valid IP addresses and test each one in turn.
4. Ping each address on your network and test any that respond.
By test I mean try to connect with Putty/SSH. Use the Username of root and password of raspberry to login.

(I'm working on submitting an enhancement to make this easier).

lurch, 2015-03-12:
Thanks to @procount for all his great advice :-)
Just to add to what he's said, IIRC the `silentinstall` option https://github.com/raspberrypi/noobs#how-to-automatically-install-an-os was added as a means of using NOOBS on headless Pis.

And if you have a smartphone, I've seen 'Fing' recommended as a way to discover your Pi's IP address http://raspi.tv/2014/top-ten-android-apps-for-use-with-raspberry-pi

Basilmir, 2015-03-13:
On version 1.4.0 (latest one at the time of posting this) the default content of recovery.cmdline is "runinstaller quiet vt.cur_default=1 elevator=deadline" and I added "vncinstall" making it "runinstaller quiet vt.cur_default=1 elevator=deadline vncinstall". Problem is I'm getting some weird error:

ip: SIOCGIFFLAGS: No such device

then nothing...

procount, 2015-03-13:
There is no problem. It is working fine. You just need to understand how to access it.

Have you now connected a monitor to the RPI?
The message you describe is typically displayed on the monitor when you add vncinstall, so it proves you have vnc running as the display for NOOBS.. You will not see anything else on the monitor at this time (all output is going to VNC instead).

You now need to get your laptop out and connect to your RPI over the network using a vnc client like tightvnc. You need to know the network IP of the RPI in order to connect to it (see previous).

When you have successfully connected to the RPI over VNC you will see the noobs screen on your laptop in the VNC client. You will then be able to install an OS like Raspbian. At the end of the installation the RPI will reboot into NOOBS again. You will need to reconnect with VNC to see the NOOBS screen. Depending on how many OS you install, NOOBS may boot that OS immediately or give you a boot selection dialog to choose which OS to boot. When NOOBs boots into that OS, you will then have to connect to it via SSH if you don't have a monitor/keyboard attached.

Basilmir, 2015-03-20:
It worked. Had a little more time to experiment and everything went as planned.

However there is one suggestion I would like to make. It's very cumbersome to find the IP of the Pi if you are a NOOB :) so it would be great to include Bonjour/mDNS (avahi or other) in this setup mode (or as standard in Raspbian) so you can call the device via 'ssh raspberrypi.local'. 

This is very convenient for the end user since Apple routers don't show the client IP addresses (not all of them, just the wireless clients) and other routers have complex interfaces to figure this out from.

Imagine copying NOOBS to a card and powering the Pi up. Connect the cable. Now you can use Screen Sharing (an OS X system app) or your favorite VNC client to connect to raspberrypi.local and set it up remotely. Wow great!!!

I think it would be in keeping with the NOOBS philosophy of making things accessible to beginners.

lurch, 2015-03-20:
> I think it would be in keeping with the NOOBS philosophy of making things accessible to beginners.

I'm not convinced I'm afraid... if you're a total beginner, then you'll just plug a keyboard and screen directly into your Pi. If you're wanting to use your Pi headless, then you should either use the `silentinstall` NOOBS option already mentioned, or just install Raspbian directly to your SD card.

IMHO wanting to use VNC to control NOOBS counts as "more than beginner" ;-)
(and as @procount already mentioned, NOOBS is currently "output to directly-attached screen" OR "output over VNC" - you can't do both at once)

Closing this issue for now, but feel free to add more comments.

closed 2015-03-20 by lurch

Basilmir, 2015-03-20:
Yes, yes! You are right about vncinstall. My overzealous suggestion wasn't to replace the way you install NOOBS. It works! Don't fix it! Vncinstall is possible, no fix required!

My suggestion was mainly about adding Bonjour/mDNS (avahi or other) so you can have an easier time finding your Pi on the network, during, after the initial install or maybe even in Raspbian. It would actually give meaning to setting a hostname, besides seeing it in bash.

Thanks for taking the time to consider this.

lurch, 2015-03-20:
Given that vncinstall isn't an officially-supported way of using NOOBS anyway, it's unlikely that I'll also be adding avahi. Unless @procount feels like adding it to his NOOBS-with-SSH branch? ;-)

To request extra features in Raspbian (which is an entirely separate project from NOOBS) please see https://github.com/asb/spindle/issues or the Raspberry Pi Forums.

procount, 2015-03-20:
Sounds like a challenge!

harpaul, 2016-03-22:
I started today with RPI 3, NOOBS 1.9, trying above method for remote installation via the vncinstall switch. Unfortunately, I do not get the GUI:
- OSX Screen sharing does not connect (it tries for several minutes)
- VNC Viewer connects, but only a black screen.
Have no HDMI monitor, so I´m stuck at this point… any ideas…?

procount, 2016-03-22:
Works fine for me on the 3B. I use Tightvnc on Windows 7.I don't have a Mac, so can't help you there. Have you tried specifying the port number as well, e.g. 192.168.2.1:5900 (replace my example IP address with your own)

maxnet, 2016-03-22:
>VNC Viewer connects, but only a black screen.

If that is RealVNC look for a fullcolour option.
See https://github.com/raspberrypi/noobs/issues/124

wazzaarr, 2016-07-14:
This is for Mac users
[image]

Traceman, 2016-07-14:
I say YES!

I know I could get in. SSH is available as well as third party software solutions. Or, dropbox, box, or skydrive an image he can put on his SD card

 

Hope this makes sense….

 

Tracy Verlin

 

 <https://cloud.githubusercontent.com/assets/20452658/16829515/234cceae-49dc-11e6-9551-6ffb6c280856.png>

kroonwijk, 2016-08-31:
This made it work from Windows using RealVNC:
[image]