raspberrypi/noobs #253: Porting NOOBS on other plafforms

ndechesne, 2015-03-25:
hi,

i am potentially interested in reusing NOOBS on other platforms, not RPi. I like the features of this tool, so I have been studying the source code, and i don't see major blockers to port to another platform (i could be wrong...) there are quite a few things hardcoded and specific to RPi, but I believe 'platform' configurations could be isolated in a decidated class/config file, at least at first glance.

i am writting this issue mainly because:
1. I'd like to know if anyone has experience with NOOBS on something else than RPi
2. I want to understand if adding other platforms is something this project would be interested (e.g. would you accept such patches?)

thanks

lurch, 2015-03-27:
I think this is one for @ghollingworth to answer...

ghollingworth, 2015-04-02:
It depends on what you want to change, go ahead and fork the code but we'd be unlikely to accept patches for other hardware here.

Gordon

closed 2018-01-30 by XECDesign