raspberrypi/noobs #254: NOOBS 1.4 (full)- Available: 0MB - Can't install software

crhea55906, 2015-04-05:
Same symptoms as #174, but I didn't want to hijack that thread.
I'm new to RPi, but experienced Linux user. 
Bought a new RP2 from MCM and with it, their 8GB Micro SD card preprogrammed with the OS (part 83-16536).  No joy. Figuring the OS must have been pre-RP2, I set out to reprogram it with NOOB as a recommended starting point. NOOBS_v1_4_0.zip MD5SUM: 176d105b17326c090df42a8971383694

BTW: My set-up works, as I have successfully loaded Raspberian onto another SD card.

Two issues:

1. Zip file has files in archive that overwrite other files. This causes non-boot. (I think there are 3.)
2. The failing repartition issue discussed in #174.

Script started on Sun 05 Apr 2015 09:29:15 AM CDT
CentOS 6 Linux: fdisk -cu /dev/sdd

Command (m for help): p

Disk /dev/sdd: 7892 MB, 7892631552 bytes
183 heads, 5 sectors/track, 16847 cylinders, total 15415296 sectors
Units = sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disk identifier: 0x0009fdca

   Device Boot      Start         End      Blocks   Id  System
/dev/sdd1            8192     2000000      995904+   c  W95 FAT32 (LBA)

Command (m for help): d
Selected partition 1

Command (m for help): p

Disk /dev/sdd: 7892 MB, 7892631552 bytes
183 heads, 5 sectors/track, 16847 cylinders, total 15415296 sectors
Units = sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disk identifier: 0x0009fdca

   Device Boot      Start         End      Blocks   Id  System

Command (m for help): n
Command action
   e   extended
   p   primary partition (1-4)
p
Partition number (1-4): 1
First sector (2048-15415295, default 2048): 
Using default value 2048
Last sector, +sectors or +size{K,M,G} (2048-15415295, default 15415295): 
Using default value 15415295

Command (m for help): p

Disk /dev/sdd: 7892 MB, 7892631552 bytes
183 heads, 5 sectors/track, 16847 cylinders, total 15415296 sectors
Units = sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disk identifier: 0x0009fdca

   Device Boot      Start         End      Blocks   Id  System
/dev/sdd1            2048    15415295     7706624   83  Linux

Command (m for help): t
Selected partition 1
Hex code (type L to list codes): c
Changed system type of partition 1 to c (W95 FAT32 (LBA))

Command (m for help): p

Disk /dev/sdd: 7892 MB, 7892631552 bytes
183 heads, 5 sectors/track, 16847 cylinders, total 15415296 sectors
Units = sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disk identifier: 0x0009fdca

   Device Boot      Start         End      Blocks   Id  System
/dev/sdd1            2048    15415295     7706624    c  W95 FAT32 (LBA)

Command (m for help): w
The partition table has been altered!

Calling ioctl() to re-read partition table.
WARNING: If you have created or modified any DOS 6.x
partitions, please see the fdisk manual page for additional
information.
Syncing disks.
CentOS 6 Linux: mkfs.vfat -n SOMETHING -F 32 /dev/sdd1
mkfs.vfat 3.0.9 (31 Jan 2010)
CentOS 6 Linux: mount -t msdos /dev/sdd1 /mnt
CentOS 6 Linux: cd /mnt
CentOS 6 Linux: unzip ~crhea/Downloads/NOOBS_v1_4_0.zip 
Archive:  /home/crhea/Downloads/NOOBS_v1_4_0.zip
  inflating: bootcode.bin            
  inflating: BUILD-DATA              
   creating: defaults/
   creating: defaults/slides/
  inflating: defaults/slides/A.png   
  inflating: INSTRUCTIONS-README.txt  
   creating: os/
   creating: os/Raspbian/
  inflating: os/Raspbian/root.tar.xz  
  inflating: os/Raspbian/boot.tar.xz  
  inflating: os/Raspbian/partition_setup.sh  
 extracting: os/Raspbian/Raspbian.png  
   creating: os/Raspbian/slides_vga/
  inflating: os/Raspbian/slides_vga/G.png  
  inflating: os/Raspbian/slides_vga/B.png  
  inflating: os/Raspbian/slides_vga/C.png  
  inflating: os/Raspbian/slides_vga/D.png  
  inflating: os/Raspbian/slides_vga/E.png  
  inflating: os/Raspbian/slides_vga/F.png  
  inflating: os/Raspbian/slides_vga/A.png  
  inflating: os/Raspbian/partitions.json  
  inflating: os/Raspbian/flavours.json  
replace os/Raspbian/Raspbian_-_Boot_to_Scratch.png? [y]es, [n]o, [A]ll, [N]one, 
[r]ename: A
 extracting: os/Raspbian/Raspbian_-_Boot_to_Scratch.png  
  inflating: os/Raspbian/os.json     
  inflating: os/Raspbian/partitions.json~  
  inflating: os/Raspbian/release_notes.txt  
   creating: os/Data_Partition/
 extracting: os/Data_Partition/Data_Partition.png  
  inflating: os/Data_Partition/partitions.json  
  inflating: os/Data_Partition/os.json  
 extracting: os/Data_Partition/data.tar.xz  
  inflating: recovery7.img           
 extracting: recovery.cmdline        
  inflating: recovery.elf            
 extracting: RECOVERY_FILES_DO_NOT_EDIT  
  inflating: recovery.img            
  inflating: recovery.rfs            
  inflating: riscos-boot.bin         
CentOS 6 Linux: cd
CentOS 6 Linux: umount /mnt
CentOS 6 Linux: sync
CentOS 6 Linux: 
CentOS 6 Linux: 
CentOS 6 Linux: ##############################
CentOS 6 Linux: ##  The above process gives you a rainbow screen.
CentOS 6 Linux: ##############################
CentOS 6 Linux: 
CentOS 6 Linux: 
CentOS 6 Linux: # Doing it again, saying "N" to the zip overwrite question:
CentOS 6 Linux: 
CentOS 6 Linux: fdisk -cu /dev/sdd

Command (m for help): p

Disk /dev/sdd: 7892 MB, 7892631552 bytes
59 heads, 1 sectors/track, 261276 cylinders, total 15415296 sectors
Units = sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disk identifier: 0x0009fdca

   Device Boot      Start         End      Blocks   Id  System
/dev/sdd1            2048    15415295     7706624    c  W95 FAT32 (LBA)

Command (m for help): q

CentOS 6 Linux: mkfs.vfat -n SOMETHING -F 32 /dev/sdd1
mkfs.vfat 3.0.9 (31 Jan 2010)
CentOS 6 Linux: mount -t msdos /dev/sdd1 /mnt
CentOS 6 Linux: cd /mnt
CentOS 6 Linux: unzip ~crhea/Downloads/NOOBS_v1_4_0.zip 
Archive:  /home/crhea/Downloads/NOOBS_v1_4_0.zip
  inflating: bootcode.bin            
  inflating: BUILD-DATA              
   creating: defaults/
   creating: defaults/slides/
  inflating: defaults/slides/A.png   
  inflating: INSTRUCTIONS-README.txt  
   creating: os/
   creating: os/Raspbian/
  inflating: os/Raspbian/root.tar.xz  
  inflating: os/Raspbian/boot.tar.xz  
  inflating: os/Raspbian/partition_setup.sh  
 extracting: os/Raspbian/Raspbian.png  
   creating: os/Raspbian/slides_vga/
  inflating: os/Raspbian/slides_vga/G.png  
  inflating: os/Raspbian/slides_vga/B.png  
  inflating: os/Raspbian/slides_vga/C.png  
  inflating: os/Raspbian/slides_vga/D.png  
  inflating: os/Raspbian/slides_vga/E.png  
  inflating: os/Raspbian/slides_vga/F.png  
  inflating: os/Raspbian/slides_vga/A.png  
  inflating: os/Raspbian/partitions.json  
  inflating: os/Raspbian/flavours.json  
replace os/Raspbian/Raspbian_-_Boot_to_Scratch.png? [y]es, [n]o, [A]ll, [N]one, 
[r]ename: N
  inflating: os/Raspbian/os.json     
  inflating: os/Raspbian/release_notes.txt  
   creating: os/Data_Partition/
 extracting: os/Data_Partition/Data_Partition.png  
  inflating: os/Data_Partition/partitions.json  
  inflating: os/Data_Partition/os.json  
 extracting: os/Data_Partition/data.tar.xz  
  inflating: recovery7.img           
 extracting: recovery.cmdline        
  inflating: recovery.elf            
 extracting: RECOVERY_FILES_DO_NOT_EDIT  
  inflating: recovery.rfs            
  inflating: riscos-boot.bin         
CentOS 6 Linux: cd
CentOS 6 Linux: umount /mnt
CentOS 6 Linux: sync
CentOS 6 Linux: exit

[image]
[image]

crhea55906, 2015-04-05:
dd if=/dev/sdd of=/tmp/first25 bs=1M count=25
25+0 records in
25+0 records out
26214400 bytes (26 MB) copied, 1.51417 s, 17.3 MB/s

fdisk -l /dev/sdd

Disk /dev/sdd: 7892 MB, 7892631552 bytes
59 heads, 1 sectors/track, 261276 cylinders
Units = cylinders of 59 * 512 = 30208 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disk identifier: 0x0009fdca

   Device Boot      Start         End      Blocks   Id  System
/dev/sdd1              35      261277     7706624    c  W95 FAT32 (LBA)

https://www.dropbox.com/s/5ikfgya9klcrldv/first25?dl=0

lurch, 2015-04-05:
> Zip file has files in archive that overwrite other files.

I've never seen that error before, I wonder if the unzip command in CentOS 6 is doing something 'funny'? Do you get the same overwriting error if you try extracting the zipfile to a (empty) folder on your harddrive, rather than directly to the SD card?
Could you try formatting and creating your NOOBS SD card on a different distro or OS?

P.S. Rather than trying to verify your downloaded file with md5sum, it'd be better to use sha1sum, as that's the checksum that's already displayed on http://www.raspberrypi.org/downloads/  ;-)

crhea55906, 2015-04-06:
Figured out this issue. You (Lurch) never see it because you do it "the right way" (the way you intended). The key is how various Linux distributions handle mounting a VFAT file system. All of the RedHat derivatives understand the difference between "mount -t msdos" and "mount -t vfat". The former folds files into 8.3 namespace, the latter uses long file names.

CentOS 6 zip is behaving.... the issue comes from files that are fine under Unix, but collide when stuffed into 8.3 name space on the SD card.  Unzipping to the card directly just makes it obvious.
I had mounted the SD card as type "msdos", which enforced 8.3 namespace.

Files that are fine in long namespace that collide in 8.3:
os/Raspbian/partitions.json  and os/Raspbian/partitions.json~
recovery7.img and recovery.img
os/Raspbian/Raspbian.png and os/Raspbian/Raspbian_-_Boot_to_Scratch.png

If you search for "recovery_7.img" in the page http://www.raspberrypi.org/forums/viewtopic.php?f=91&t=83372&p=718192 (In the posting, the file name is typoed-- really is recovery7.img), you'll see others have tripped over this (with no solution).

lurch, 2015-04-06:
Ahhhh, so I presume that by starting again and changing "mount -t msdos" to "mount -t vfat", you were then able to get your SD card booting successfully?

Closing this issue as invalid, but feel free to add extra follow-up comments.

crhea55906, 2015-04-06:
I still have the re-partition issue.  Yes, machine boots, no, I cannot install any OS as I get the Available Space: 0MB.

lurch, 2015-04-07:
Hmmm, okay. So in order to track down exactly where the problem lies, can you try all the above with a different SD card? Does NOOBS Lite have the identical problem? Are you able to perform the formatting step (mkfs.vfat) on a different distro or OS? Do you also have a Pi1, or only a Pi2?

crhea55906, 2015-04-07:
I must apologize....  It appears I didn't fully re-test after figuring out the mount -t msdos / mount -t vfat
issue. I rebuilt from scratch this morning (same SD card) to try in a RPi1B and it worked. Thinking I was onto something, I rebuilt and tried it in my RPi2-- again worked.

The symptoms only return after mounting the SD card incorrectly as type "msdos" to load it.
I'm good... Thanks!

closed 2015-04-07 by crhea55906

lurch, 2015-04-07:
Glad to hear you've managed to get it working :)