raspberrypi/noobs #255: busybox console in NOOBS?

beta-tester, 2015-04-07:
does NOOBS has an option to launch a full featured busybox console?
if not, it would be a very nice option to have for many maintain purposes...
e.g. restoring single partitions by hand.

lurch, 2015-04-07:
Yup! :-)
https://github.com/raspberrypi/noobs#how-to-boot-into-safe-mode

closed 2015-04-07 by lurch

lurch, 2015-04-07:
...but let me know if you meant something different.

beta-tester, 2015-04-07:
oops, sorry, i didn't know that this is already possible.
yes, that is, what i was looking for.
but it is not able to enable by an menu option, is it right?
(only able to enable via GPIO and recovery.cmdline)

procount, 2015-04-07:
Alternatively, press Ctrl-Alt-F2 to get to the rescue shell.

beta-tester, 2015-04-07:
oh man, you are my hero... many thanks. :+1: :)

procount, 2015-04-07:
Hey @beta-tester, if you fancy doing some "beta-testing" have a look at https://github.com/procount/pinn/ re: some of your previous requests. README.md is not quite up to date, but look at https://github.com/procount/pinn/blob/master/newfeatures.md for all new additions.
(Warning: This is Beta quality software - use at your own risk).

NOOBS will probably not see your USB stick, because it does not have USB support built in, but PINN does.

beta-tester, 2015-04-07:
ah, user is **root** and password is **raspberry** ... :)

@procount: the PINN project looks interesting.
USB memory stick and network support, very nice features.
is it also possible to boot into a partition on the USB stick (and install an OS to an existing partition on the USB stick)?
maybe i will give it a try bit later...
(of cause at my own risk - as everything at every time)

lurch, 2015-04-07:
In theory the root partition could be on a USB stick (you'd just need to change which root partition the cmdline.txt points to), but the boot partition *always* has to be on the SD card. Technical details [here](https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained)

procount, 2015-04-07:
The USB support in PINN is only to allow the stick to be a SOURCE of compressed OSes, not a destination.
Having installed an OS to the SD card, you could manually move the ROOT part to the SD card and do the fixups to cmdline.txt and fstab. (I did this onto a usb HDD).
But I can't see NOOBS or PINN doing this for you. Making the USB stick be a source of compressed OSes and a destination for where to install the root part to would require a bit of thinking about! One or the other would be simpler.

procount, 2015-04-07:
I should re-iterate my credits to robbi5 (USB) and jumaka (WIFI) for their mods - I just integrated them together with my own mods.