raspberrypi/noobs #257: NOOBS installer does not recognize HID (Keyboard/Mouse)

phillipremaker, 2015-04-15:
Using a Raspberry Pi 2, the following HID was not recognized:

ID 04fc:05d8 Sunplus Technology Co., Ltd Wireless keyboard/mouse

Works just fine on the RASPIAN after install, but not in the NOOBS environment.

lurch, 2015-05-04:
As a temporary workaround until I get more time to investigate this, do you have a different keyboard or mouse you could use?

phillipremaker, 2015-05-04:
Yes.  I did install successfully with an alternate wired keyboard and once
on Raspian I switched back and it worked fine.

lurch, 2015-05-05:
Can you provide more details about this particular keyboard? A link to an Amazon.co.uk product listing would be ideal ;-)

procount, 2015-05-05:
This looks like another issue with the same chipset in Ubuntu (although from 2009)
[http://ubuntuforums.org/archive/index.php/t-1080161.html](http://ubuntuforums.org/archive/index.php/t-1080161.html)

phillipremaker, 2015-05-05:
This is an Inland Wireless Keyboard/Mouse combo that I got from Fry's years
ago.  I think  this specific model is discontinued (70119).  Mine is the
one with a larger wireless dongle (not the "nano" one, that currently
ships).

This, for example, is the NEWER one:
http://www.amazon.com/Inland-Wireless-Optical-Keyboard-70119/dp/B0026PTO1I

This is a photo of the old one that I have.
http://www.inlandus.com/inlandfaq/docs/instructions/70119_Instructions.pdf

lurch, 2015-05-05:
I've just taken a punt on http://www.amazon.co.uk/gp/product/B00H6LA632/ - in the middle photo it seems to be using a similar "non nano" receiver dongle.

phillipremaker, 2015-05-05:
Yes, based on other data in a Google search  about the Rappo, it looks like
that is just a relabeled SunPlus unit, just like the Inland.  It looks
physically identical to the keyboard/mouse combo I have.

MobiusTesseract, 2015-06-05:
I do have the same problem with a completely diffirently looking keyb/mouse combo "Silvercrest MTS2218" (bought from LIDL in Europe). It uses the same non-nano adapter and appears as ID 04fc:05d8 Sunplus Technology Co., Ltd as well. Neither mouse or keyboard respond in NOOBS.

adamst85, 2015-12-10:
I also have the same problem with yet a different keyboard/mouse combo. ID is 1241:f767 and it works in both Raspbian and Openelec no issues (it is even on the RPi USB keyboards page http://elinux.org/RPi_USB_Keyboards). I have checked on OpenElec and it is using the usbhid driver so it should work no problem.

adamst85, 2016-01-15:
I have forked Noobs and added support for as many wireless keyboards as I could find (and also added the PS3 controller and Wii mote for good measure :). Fork is here: https://github.com/adamst85/noobs and compiled result is available here: http://www.megafileupload.com/isZr/Noobs-keyboard-support.zip

tomchiverton, 2016-03-03:
NOOBS also does not seem to work with the Pi 3 and a Rii Mini i8.

maxnet, 2016-03-03:
Bluetooth ones are currently not supported.
But Sunplus and the like should be fixed by 6894dd251e5cf86c06b3a778a53a8944c3817208

closed 2018-03-07 by XECDesign