raspberrypi/noobs #258: 0 Disk Space Available.

dempsone, 2015-04-22:
Hi All,

Wondering you guys can help.

I have an 8GB SD card and a Raspberry Pi 2 B+

When i boot into NOOBS the first time and try to install Rasbian it says there is 0 disk space available.
I have read through similar issues other users were having and followed the advise provided by you guys but have had no luck :<

For example after formatting the SD card correctly using SDFormatting tool. I tried to resize the partition.

 resize 1 8192s 143M

I get  Error: Unable to satisfy all constraints on the partition.

Is it because my SD card is a few meg smaller? should be 8GB but it was bought off ebay?

p shows

Number Start End Size Type File System
1 4194KB 7904MB 7900MB Primary fat32

I appreciate your help guys.

lurch, 2015-04-22:
SD cards from different manufacturers come out at different sizes - so being a several meg under 8GB is nothing unusual. If you bought it off ebay it's worth checking it's not a 'fake' SD card using e.g. [H2testw](https://sosfakeflash.wordpress.com/h2testw/).

`parted` (the formatting tool used by NOOBS) can sometimes be a bit "picky", and it sounds like you're unlucky enough to have a card of the exact size that parted is unhappy about :-(
You could try copying some "dummy files" onto the card first (maybe 20MB or so - perhaps a copy of `NOOBS_lite_v1_4.zip`) before booting NOOBS and perhaps that will nudge around the partition sizes enough for parted to stop complaining...?
If that doesn't work, do you have a regular linux computer (or a Windows PC you can boot from a Linux LiveCD/LiveUSB) that you could use to manually partition the card?

dempsone, 2015-04-23:
I noticed the Make/Model of the SD card is "000000" Sounds pretty generic to me?

Ill use H2testw and let you know how I go, I have a feel that its not 8gb at all.

lurch, 2015-05-04:
Any updates, @dempsone ?

XECDesign, 2017-11-03:
No response

closed 2017-11-03 by XECDesign