raspberrypi/noobs #259: Leave config.txt options to the maintainer

samnazarko, 2015-04-29:
I think that these options should be left to the maintainer. On OSMC some of these values are already overlapping.

    # NOOBS Auto-generated Settings:
    hdmi_force_hotplug=1
    config_hdmi_boost=4
    overscan_left=24
    overscan_right=24
    overscan_top=16
    overscan_bottom=16
    disable_overscan=0

lurch, 2015-05-04:
@samnazarko I'm not sure why these (specific) values are being set, nor what impact *not* setting them would have on other NOOBS-installed OSes. Any ideas @maxnet , @popcornmix , @ghollingworth or @asb ?

What if I modified NOOBS so that it didn't append any overlapping options that were already present in the maintainer-supplied config.txt? (and didn't append any `overscan_*` settings if the existing config.txt said `disable_overscan=1`)

samnazarko, 2015-05-04:
Our desired settings are:

echo "arm_freq=850" >> config.txt
echo "core_freq=375" >> config.txt
echo "gpu_mem_256=112" >> config.txt
echo "gpu_mem_512=144" >> config.txt
echo "hdmi_ignore_cec_init=1" >> config.txt
echo "disable_overscan=1" >> config.txt
echo "start_x=1" >> config.txt

So disable_overscan seems to be the only overlapping one. Another option would be to add your custom settings before the maintainer's. Any customised settings will then get overwritten.

popcornmix, 2015-05-05:
How about an option in the OS's json file to request config.txt is untouched by NOOBS?

lurch, 2015-05-05:
That's an even better idea, thanks @popcornmix :-)
(of course this new option would then mean it's up to the installed OS to decide whether to display over composite or HDMI, rather than 'inheriting' the output mode from NOOBS)

samnazarko, 2015-05-05:
Works for me.

maxnet, 2015-05-05:
>@samnazarko I'm not sure why these (specific) values are being set, nor what impact not setting them would have on other NOOBS-installed OSes. Any ideas @maxnet , @popcornmix , @ghollingworth or @asb ?

Don't know the reasoning behind the exact settings used now.
Recall in the original code only the output mode (hdmi preferred/safe mode/sdtv) was set.