raspberrypi/noobs #260: NOOBS download link gives 404 error

electronicsguy, 2015-05-11:
the download link for Noobs latest gives me 404 error. Only once (out of 20 times) did it succeed:

http://downloads.raspberrypi.org/NOOBS_latest

lurch, 2015-05-11:
It's always worked fine for me... (including just now)

That link is a redirect to another URL, which it's possible your browser isn't handling properly? Try using a different browser to download the file, and if that doesn't work try this direct link: http://director.downloads.raspberrypi.org/NOOBS/images/NOOBS-2015-02-18/NOOBS_v1_4_0.zip
If it still doesn't work, try downloading NOOBS Lite instead (perhaps your ISP has some kind of restriction on downloading big files?)

Ruffio, 2015-06-17:
@electronicsguy is this still an issue?

lurch, 2015-06-18:
Closing due to lack of inactivity. Feel free to add further comments if you're still having problems.

closed 2015-06-18 by lurch

electronicsguy, 2015-06-18:
No, its working fine now.