raspberrypi/noobs #264: Compute Module - problems with inializing MMC on warm boot

kuba-moo, 2015-06-09:
I'm trying to use NOOBS on RPi Compute Module to provide a minimal fallback OS if someone screws up configuration of the main one in a remote access scenario. I'm having trouble to get NOOBS to work reliably, though. If I power the unit up too quickly after it was powered down it will not boot displaying error:

`mmc0: error -110 whilest initializing MMC card`

I can reproduce it 100% with the following procedure: remove power, wait 15 seconds, remove the modules from dev kit, wait 5 seconds, put the module back in and power it up.  When the error occurs NOOBS gets confused and I'm unable to boot any of my OS-es. The error also does not go away if I leave the unit powered (mmc error is printed couple times a second until I reboot again). 

However, if I wait a few minutes after removing power instead of couple of seconds, MMC memory is initialized just fine and everything works.

lurch, 2015-06-10:
Strange! I'll try to have a go at replicating this before the weekend.

Can you confirm what version of NOOBS you're using (paste the contents of the `BUILD-DATA` file), and what `cat /proc/cpuinfo` says on your ComputeModule?

Pinging @ghollingworth too, as he has more ComputeModule experience than me.

kuba-moo, 2015-06-10:
I'm testing with v1.4 (lite):
```
rpi-userland Git master @ c5341f09dc686c17966369485f7dd27f58dc081c
rpi-firmware Git master @ 8aca5762f984f6decbeda294cee8418966c3d8d3
rpi-linux Git rpi-3.18.y @ b630db9dcb5c73367050a672823047b0466b16b9
```
cpuinfo:
```
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
Features	: swp half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2708
Revision	: 0014
Serial		: 00000000bd784c57
```
I did try to use firmware files from Raspbian and firmware repo but no success...

kuba-moo, 2015-06-10:
Actually it seems @ghollingworth [implemented](https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=96646) something like this for the Slice product. Are the sources of Slice available anywhere?

ghollingworth, 2015-07-31:
I believe there was a bug in one of the linux versions which broke eMMC reading on linux...  I'd suggest you pulling up the linux version to the head of the 3.18 tree where it should be fixed

Gordon

XECDesign, 2018-03-07:
Should be fixed. Please re-open if this is still an issue.

closed 2018-03-07 by XECDesign