raspberrypi/noobs #265: Minor filesystem issues cause noobs failure.

plugwash, 2015-06-14:
I was at a jam recently and a newbie was having problems with noobs. The resize was complaining about a minor filesystem issue (free space inconsistent) and hence refusing to resize the partition and hence unable to install any OS.

Putting the card in another linux box and running fsck with the "auto fix" option fixed the issue but I wouldn't expect newbies to be able to do that themselves.

IMO it would make sense for noobs to do an "auto-fix" fsck run before doing the partition resize.

lurch, 2015-06-18:
Unfortunately this comes under the same category as #174 - without a reproducible test-case, I can't test that the proposed fix always works as expected.

If you have a way in which this **can** be reproduced, that would be fantastic... :-)

plugwash, 2015-06-21:
Using information from https://staff.washington.edu/dittrich/misc/fatgen103.pdf I was able to construct the following recipie to make an image which I believe will suffer from the issue (note: this requires a recent losetup, wheezy's version is too old, I think jessie's is sufficient)

truncate -s 4G testimg
parted testimg
unit b                                                           
mklabel msdos
mkpart primary fat32 4194304 100%
print 
quit
loopdev=`losetup -P -f --show testimg`
echo loopdev
mkdosfs /dev/loop0p1
printf '\x40' | dd of=${loopdev}p1 bs=1 seek=1000
losetup -d $loopdev

plugwash, 2015-06-21:
ok i've posted the SD card image I created with the process above at https://plugwash.raspbian.org/badfreespace.bz2

Try writing that to a SD card, then copying the noobs files and booting it in a pi and see what happens.

lurch, 2015-06-23:
Thanks @plugwash !
I'm busy all this week, but I'll take a look at this next week.

Androbin, 2018-02-26:
Has this issue been resolved yet?

XECDesign, 2018-02-26:
I don't think so.