raspberrypi/noobs #266: NOOBS Online Help link does not load (Lite)

simonsmall, 2015-06-15:
Using NOOBS Lite v1_4 downloaded 14 June 2015, clicking the Online Help button opens the Arora browser to load page "https://www.raspberrypi.org/help". This page does not seem to load (I gave it a while, then stopped it). Other info:
* Entering any http:// address opens, but https:// does not
* The address given in the 'recovery/config.h' file uses http:// but the site replaces it with https://
* The documentation suggests that this goes to the Forum (also uses https:// )
* I have not checked with the full NOOBS system

lurch, 2015-06-19:
Confirmed. The problem is that the Raspberry Pi website has recently been modified to silently redirect all http:// requests to their https:// equivalent, and it seems like the Arora browser built into NOOBS (it's the same in both NOOBS and NOOBS Lite) isn't able to cope with these https:// URLs, which is something I'll need to investigate further...

> The documentation suggests that this goes to the Forum (also uses https:// )

Well spotted, I've now updated the documentation to match what NOOBS actually does :-)

maxnet, 2015-06-19:
Qt's (Open)SSL support has been disabled in the Buildroot config in the past.

Before the Raspberry site switched over to using ssl completely, it was already including some Javascript code from a third-party ssl site. The code was not necessary for the site to function (recall it was some statistics/analytics thing) but it did generated big fat certificate warning popups in Arora, because the Pi does not have a RTC, thought it was 1970 and the certificate's start date said it was not valid in 1970 yet...

So if you want to re-enable ssl support, you will need to add code to set the correct time for it to work properely.
You can probably grab the time from the repo server, webservers tend to send the server time in the http headers.
Or use ntp, but that may not pass firewalls used in corporate networks.

lurch, 2015-06-20:
Thanks for the info Floris :-)

`wget -q -O /dev/null -S http://downloads.raspberrypi.org/os_list_v2.json 2>&1 | grep Date:` on my desktop Linux PC indeed shows that the webserver sends the info I need (for when I add OpenSSL back into NOOBS).

commit 78b8e4dc (2015-11-26): Enable SSL support

- Enable SSL support in Qt
- Enable installation of CA certificates
- Set system clock to the date/time of the repository web server ("Date" HTTP response header)

Closes #266
closed 2015-11-26 by maxnet