raspberrypi/noobs #268: How is syslog output printed on TTY in NOOBS?

ardrabczyk, 2015-07-02:
When I log in as root in TTY in NOOBS the following text is printed:

    login[510]: root login on 'tty2'

I know it can be achieved by modifying syslog.conf but there is no syslog.conf and syslogd is not running in NOOBS. I read Busybox source code and the only place when this text is printed is loginutils/login.c but they use syslog():

	if (pw->pw_uid == 0)
		syslog(LOG_INFO, "root login%s", fromhost);

I wonder then where does this text come from?

lurch, 2015-07-02:
What is it exactly that you're trying to achieve?

ardrabczyk, 2015-07-02:
?? Nothing, it's a question. I asked the same on Raspberry forum: https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=114606

lurch, 2015-07-26:
I'm afraid I'm too busy to investigate a minor question like this myself, so you'll either have to trace through the sourcecode yourself, or you could try asking on the BusyBox mailing list?

closed 2015-07-26 by lurch

ardrabczyk, 2015-07-26:
I already did, it's a Busybox thing: http://lists.busybox.net/pipermail/busybox/2015-July/083078.html. I thought it may be specific to NOOBS but it's not.