raspberrypi/noobs #269: A few questions on how NOOBS handles multiboot...

kusti8, 2015-07-05:
I want to rewrite NOOBS, but made for advanced users with multiboot capabilities and being able to change the partition sizes etc.

My problem is, I can't understand how NOOBS deals with multiboot. I've looked through the source code and stumbled across multiimagewrite.cpp (or something like that). I've found out that there is a /sys file where it writes to set the boot up partition, but how it installs those OS is the question. It uses tarballs, although I would like to use .img files to make it easier to add new OSs.

So it creates a extended partition and in that it creates the primary partitions for the OS (?), but how would I implement that or how does NOOBS implement that for .img files. Can dd start at a partition?

I've looked through the wiki, but couldn't find the answers I needed.

Any help is appreciated!

HanSooloo, 2015-07-24:
You could probably use 'kpartx' to extract the boot and root partition contents from an .img file and just untar. 

That's what I did to do a side-by-side install of MINIBIAN together with Raspbian.

kusti8, 2015-07-24:
I already found out how to do it. I decided to just go with tar files. I plan on implementing img files, but my OS runs in RAM, so I can't download the entire image file, mount it and copy the files. I tried skip with dd, but for some reason it didn't work, so I'll just stick with tar files for now and use img files if there is a place to store them since the SD card is not an option. 

I'll close this issue.

lurch, 2015-07-24:
Glad to hear you got it working :-)

The other advantages that tar files have over img files, is that tar files don't contain empty blocks (so are typically much smaller to download), and with a tarfile you create the partition first (at whatever size you want), format it, and then extract the tarfile contents to it. When extracting an img file it always creates a fixed-size partition, so you'd need to resize it afterwards if you wanted a different partition size.

kusti8, 2015-07-24:
Yeah. I successfully got Raspbian installed and booted. Now I'm off to try OSMC. 

My final goal is to make a version of NOOBS geared towards intermediate and advanced users to give them more control. 

Your wiki on NOOBS partitioning was very helpful. :)

kusti8, 2015-07-24:
One last thing (I don't know if I should reopen this or not, but whatever :)), I read in the wiki about RiscOS needing a special offset. I'm going to experiment some more, but one question that I have, is what the RISCOS-boot.bin file is for and what it does?

Thanks :)

lurch, 2015-07-26:
There's some in-depth info about `riscos-boot.bin` in #262 

Basically if it's not present, then NOOBS is unable to boot RISC OS.

lurch, 2015-07-26:
> My final goal is to make a version of NOOBS geared towards intermediate and advanced users to give them more control. 

If you do do that, please call it something other than "NOOBS", and make sure it's clear that it's a derivative version, and not something that's developed or supported by Raspberry Pi.
You might want to collaborate with @procount and / or @meltwater2000 as I believe they are / were working on similar ideas.

kusti8, 2015-07-26:
Interesting. I'll read that issue more in depth later. 

As for your second comment: I'll keep that in mind. :)

meltwater2000, 2015-07-26:
Hi, Yes we ended up developing the PiKitchen, it is still very much developmental (and I've not had time recently to work on it) but you can find further details here:
https://pihw.wordpress.com/guides/pi-kitchen/

lurch, 2016-02-06:
Closing issue, as I believe your questions have been answered now?

closed 2016-02-06 by lurch