raspberrypi/noobs #271: Cannot 'reboot' in Safe Mode rescueshell

HanSooloo, 2015-07-16:
**Steps to reproduce:**
1. Mount /dev/mmcblk0p1 on /mnt/p1
2. Add 'rescueshell' to /mnt/p1/recovery.cmdline
3. Reboot
4. At Safe Mode rescueshell, 'reboot'

**Observed results:**
No output.
Goes back to command prompt, as if previous command didn't do anything other than 'exit'
```
/ # reboot
/ #
/ # echo $?
0
/ #
```

**Expected results:**
System reboots.

lurch, 2015-07-16:
I suspect rescueshell is an option used by very very few users.
What's the use-case for having this work properly, rather than simply pulling and re-inserting the power?

HanSooloo, 2015-07-16:
When the Raspberry Pi is not close by, it would be easier to control via command line, rather than trying to unplug/replug. 

Higher level use case is as follows:

Before a major dist upgrade, boot into rescue shell.
Start networking. 
Mount NFS export from backup server. 
Mount desired OS partition. 
rsync from OS partition to NFS mount. 
Unmount.
Reboot. 
Do dist upgrade.

maxnet, 2015-07-16:
"reboot" normally nudges the init process to get the job done. We have a simple single file shell script as  startup script, instead of a full-fledged sysv-style init infrastructure.

Make sure nothing is mounted (umount -ar) and try the three finger salute (ctrl+alt+del) as alternative.

lurch, 2015-07-26:
I've not tested, but I doubt that the rescueshell environment supports NFS anyway?

You could always use the 'flavours' feature of NOOBS to install two side-by-side copies of Raspbian, and then boot into one Raspbian install in order to backup the other Raspbian install?
#195 shows how you can change which OS NOOBS boots into, if you're running it 'headless'.

HanSooloo, 2015-07-26:
You are right ... in the end, I installed MINIBIAN as a 2nd OS to do NFS rsync backups of the main OS, i.e., Raspbian.

As an educational exercise, I'd actually modified the recovery root and kernel to provide NFS (using buildroot menuconfig), but could not get WiFi to work (different issue).

I am still at a loss, as to why reboot/shutdown would do __absolutely nothing__ ... what kernel feature or external package does it need that it doesn't find that results in this "do nothing" state, I wonder.

maxnet, 2015-07-26:
>I am still at a loss, as to why reboot/shutdown would do absolutely nothing ... what kernel feature or 
>external package does it need that it doesn't find that results in this "do nothing" state, I wonder.

"reboot" only nudges the init process (pid 1)

For educational purposes try adding: ``trap 'echo received a signal!' SIGUSR1 SIGUSR2`` to the top of package/recovery/init, and see if that can catch the reboot request.
If so, you can add a proper shutdown sequence like "umount -ar ; reboot -f" there.

plughplover, 2015-07-29:
on a related note, the init script has the following construct:

>52 if ( [ ${SAFE_MODE#*=} -eq 1 ] && ! grep -q disablesafemode /proc/cmdline ) || grep -q rescueshell /proc/cmdline; then 
>53     sh 
>54 else 
>.....
>91     /usr/bin/recovery $RUN_INSTALLER $GPIO_TRIGGER $KEYBOARD_NO_TRIGGER $FORCE_TRIGGER $DEFAULT_KBD $DEFAULT_LANG $DEFAULT_DISPLAY $DEFAULT_PARTITION -qws 2>/tmp/debug 
>93 fi 
>97 echo Recovery application crashed 
>99 sh 

which means that 'exit' from safemode / rescueshell just gives you another shell
Suggestion: un-else-ify the 'recovery' block. i.e.

>if ...safemode || rescueshell then
>   sh
>fi
>...
>   /usr/bin/recovery...
>echo Recovery application crashed
>sh

i.e. when safemode / rescueshell, give them a chance to do whatever, 
and then continue on with the normal recovery/boot process.

procount, 2017-03-10:
A command line reboot can be affected by:
`echo b>/proc/sysrq-trigger`

HanSooloo, 2017-03-10:
The /proc suggestion above seems to be the simplest one.

closed 2017-03-10 by HanSooloo