raspberrypi/noobs #278: Apple Bluetooth Keyboard and Magic Mouse not detected in OS selection screen

aoberoi, 2015-09-08:
I'm not sure if bluetooth mice and keyboards are intended to work for the OS selection screen during boot, but I hope so.

My Apple keyboard and Magic Mouse are not being detected, or at least not during the 10 seconds in which an operating system can be selected. They work fine in Raspbian. Any ideas?

tomchiverton, 2016-03-03:
They should work. Looks like this is a duplicate of #257

procount, 2016-03-03:
Really? Since when has Noobs supported Bluetooth?

tomchiverton, 2016-03-03:
IDK. But now the Pi has Bluetooth on-board, doing so would be a good plan. Why should I have to connect a USB keyboard just to change the boot order or do the initial install ?

procount, 2016-03-03:
I agree, but adding bluetooth will bloat noobs somewhat. And you will still have to pair the keyboard with the Pi before it will communicate. And somehow store the pairing information so it will remember it for next time. It's a lot of work for little gain and I doubt the maintainers will consider it worthwhile.

tomchiverton, 2016-03-04:
Not true that it's a complicated process involving storing PINs etc
My keyboard pairs with openelec without entering a pairing PIN, only select 'connect' on it from the list of devices. 
-- 
Tom
Sent from my phone.

procount, 2016-03-04:
Well, elsewhere @maxnet said "bluetooth is not CURRENTLY supported" so maybe there is hope for you....

XECDesign, 2018-03-08:
No plans to add bluetooth support in the near future.

closed 2018-03-08 by XECDesign