raspberrypi/noobs #279: noobs does not allow setting proxies to access the internet

tps800, 2015-09-10:
NOOBS does not allow setting proxies to access internet. This might be necessary within various organisations like universities, collages, schools, companies, ...

Would therefore be quite nice if I could configure something like

ftp_proxy=http://myproxy.local:port
http_proxy=http://myproxy.local:port

to have NOOBS download OS-images from internet instead just telling me about "could not connect servers".

beta-tester, 2016-03-05:
maybe it is similar to
https://github.com/raspberrypi/noobs/issues/231
https://github.com/raspberrypi/noobs/pull/234

lurch, 2016-03-06:
Nope, it doesn't have anything to do with #231 or #234 - those are concerned with accessing a *different* `os_list.json` whereas AFAIK this issue is concerned with accessing the 'normal' `os.json` when internet access is only available via an HTTP proxy, rather than directly.

pradithya, 2017-10-19:
+1 encountered this issue while setting up windows 10 IoT core behind my company proxy.

cccaballero, 2018-03-18:
+1
I just found this issue right now, It will be really useful to be able to set a proxy configuration.