raspberrypi/noobs #282: Do most users have to format cards when setting up NOOBS?

rjsteinert, 2015-09-27:
I'm wondering if formatting an SD Card is a common barrier to NOOBS. Formatting would need to take place in the following scenarios (as far as I know):

1. Fresh SD Card did not come formatted as FAT32.
2. Reusing a SD Card that is not currently formatted FAT32.

When setting up a fresh SD Card I'm usually just unmounting it and writing an image to it so I don't usually notice the format the card comes in. As for new card vs. reusing card, there's no simple answer for that.  

Has anyone studied this?

ghollingworth, 2015-09-28:
Most SD cards now come preformatted as ex-FAT and the raspberrypi cannot read that.

lurch, 2015-10-21:
I don't know what you mean about formatting being a barrier to NOOBS? It's always clearly mentioned in the [instructions](https://www.raspberrypi.org/documentation/installation/noobs.md).

The exFAT issue Gordon mentions is documented [here](https://www.raspberrypi.org/documentation/installation/sdxc_formatting.md).

XECDesign, 2017-11-03:
Answered

closed 2017-11-03 by XECDesign