raspberrypi/noobs #283: Suggestion: Offical Touch Display support

Ponsen, 2015-09-29:
Hi!

I just got my new Touch Display and I wanted to suggest the the ability output the screen on said display. Especially when it comes to multi-boot it's a downside not being able to see anything without a HDMI display nearby.

Thank you a lot in advance!

freekert, 2015-10-01:
I agree! This would be an extremely useful feature, it would eliminate the TV from the entire process.

procount, 2015-10-01:
See my post (5th post in the thread) here 
https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=120332#p814214.

Ponsen, 2015-10-02:
Can confirm it working.

procount, 2015-11-26:
I managed to add touchscreen support to the old NOOBS kernel.
But it appears the RPF have released a new NOOBS with an up to date kernel that uses device tree and includes touchscreen support. Go to the downloads webpage. When will the source be released on github?

maxnet, 2015-11-26:
>When will the source be released on github?

It is there now.

closed 2017-11-03 by XECDesign