raspberrypi/noobs #285: NOOBS v1.4.2 (offline, Raspbian only) cannot be used with a 4GB SD card

DeeJay, 2015-10-05:
The first line of INSTRUCTIONS-README.txt says "Insert an SD card that is 4GB or greater in size into your computer."

Since the NOOBS Recovery partition will occupy almost 1GB, and the image file for 2015-09-24-raspbian-jessie.img is itself slightly larger than 4GB, it seems that a 4GB SD card will not be sufficient.

lurch, 2015-10-21:
I already mentioned this to the Raspbian maintainer, and it's *possible* that future versions of Raspbian will be made slightly smaller, to squeeze in under the 4GB barrier (which will also alleviate the unzipping problems mentioned on https://www.raspberrypi.org/blog/raspbian-jessie-is-here/#comments ).

XECDesign, 2017-11-03:
INSTRUCTIONS-README.txt has been updated since then

closed 2017-11-03 by XECDesign