raspberrypi/noobs #286: Error kernel panic not syncing vfs unable to mount root fs on unknown-block(0;0)

mixu1308, 2015-10-17:
I have tried multiple times to use noobs for multiboot but I keep getting this error:kernel panic not syncing vfs unable to mount root fs on unknown-block(0;0).
I have tried installing openelec and raspbian and both of them worked.
I can't find help anywhere on the internet about this error for NOOBS.
Any help is greatly appreciated.

popcornmix, 2015-10-18:
I think NOOBS is on the 3.18 kernel, and raspbian/openelec are on 4.1 kernel which has additional sdcard fixes. NOOBS really needs updating to a later kernel.

mixu1308, 2015-10-18:
So I should just wait?

popcornmix, 2015-10-18:
Yes. Or try a different sdcard.

mixu1308, 2015-10-21:
Ok so I tried installing noobs 1.4.2 again just to be sure and I got it to work since last time I was using 1.4.1. I then did sudo halt to shut it down, and now I get the error again :/

lurch, 2015-10-21:
Yeah - the "1.4" at the start of both "NOOBS v1.4.1" and "NOOBS v1.4.2" indicates that they're both based on the same "core version" of NOOBS, which means the kernel hasn't changed (when the kernel does change, it'll then be NOOBS v1.5.x).
So if it doesn't work on NOOBS v1.4.1 then I'd expect it to also not work on NOOBS v1.4.2

See also https://github.com/raspberrypi/noobs/issues/287#issuecomment-149877958

mixu1308, 2015-10-21:
But the thing is that it worked once and then it broke and now I get the same error again

lurch, 2015-10-21:
You were probably just "lucky" that it worked once. From what (little) I understand, this is a subtle timing problem with particular SD cards (with the newer kernel driver being more robust).

As @popcornmix says, your best bet for now is to try a different SD card. http://swag.raspberrypi.org/collections/frontpage/products/noobs-8gb-sd-card is very reliable and has never caused any problems.

procount, 2015-10-21:
This sounds very similar to https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=123446 where the rootfs could not be found due to an error in cmdline.txt.

lurch, 2015-10-21:
Similar error-message, but a different cause ;-)
https://github.com/raspberrypi/firmware/issues/397

mixu1308, 2015-10-21:
Anyway thanks for the info,  is there an ETA? Do you have any recommended micro SD cards that work with the raspberry Pi 2?

lurch, 2015-10-21:
No ETA yet.
The card I linked to earlier is actually a microSD card in a full-size SD adaptor :-) It works fine on all models of Raspberry Pi, and is in fact the card I use for all my NOOBS testing.

mixu1308, 2015-12-06:
Just wanted to say that 1.5 solved my issue

HOD42, 2017-09-02:
I had a similar issue when I swapped a card from a PiZeroW to a regular PiZero. I didn't do a clean shutdown (the system was supposed to be using tempfs for everything). Anyhow, on booting the PiZero it was failing to boot with the "Error kernel panic not syncing vfs unable to mount root fs on unknown-block(0;0)" message. I simply inserted the card in a Windows PC and renamed the /boot/cmdline.txt (effectively removing it). On reboot, it worked fine!

XECDesign, 2017-09-03:
Closing, since it was solved in 2015.

closed 2017-09-03 by XECDesign