raspberrypi/noobs #287: error 110 - transferring data - noobs cannot install raspbian on the TS32GUSDU1E Transcend 32GB Premium 300x card

saraf, 2015-10-21:
[This card](http://www.amazon.com/Transcend-MicroSDHC-Class10-Adapter-TS32GUSDU1E/dp/B00CES44EO) does not work. There is a stream of mmcblk0 error 110 transferring data errors on the console.

And now I see [this list of cards that are verified to work and not work](http://elinux.org/RPi_SD_cards)

I am surprised to see a large number of cards that do not work!

Can this be fixed?

(Note: the kernel with the Raspbian image does not have this problem. I had reported this error at https://github.com/raspberrypi/firmware/issues/473 . Maybe moving to a newer kernel could fix this?)

lurch, 2015-10-21:
I wouldn't trust that list of "cards that don't work" on the eLinux wiki - much of that information is stale and out-of-date.
It's possible a newer kernel in NOOBS will fix this problem (and hopefully also #286 and #230 ) but I'm afraid there's no ETA on that yet.

NicoHood, 2016-02-07:
I had a lot of problems with this sd card too.
At some day raspbian for pi1 fixed those issues. Then with the pi2 it took some time and its fixed again. However the new 4.4 kernel (with ne sd driver modul) causes sd corruption again.

There are still distributions which wont work with this SD card like the asterisk image.
I stopped using this sd card when it finally broke. I cannot write to it anymore. I suggest to get the money back if it was ordered from amazon. They never complained about that.

I suggest to use the samsung evo+ instead. But the current noobs works with this sd card though.

XECDesign, 2017-11-03:
If this is still a problem, it should be reported here:
https://github.com/raspberrypi/linux/issues

closed 2017-11-03 by XECDesign