raspberrypi/noobs #290: Image downloads are not signed

bgeron, 2015-11-22:
Looking at `recovery/config.h` and http://downloads.raspberrypi.org/os_list_v2.json, it seems that if users are on a compromised network, they might install an image with a virus. Most home routers are closed source, rarely updated, and are very often [exploitable](http://routersecurity.org/bugs.php). 

There are two obvious ways to defend against this: https and signatures (e.g. apt-style). The former will not work because I believe the rpi does not have an entropy source. 

This is how I understand apt implements signatures:

- The system has public keys of people who sign "good" packages in `/etc/apt/trusted.gpg`
- The "release" http://archive.ubuntu.com/ubuntu/ubuntu/dists/wily/InRelease is downloaded. It has a signature, which is checked.
- The package list http://gb.archive.ubuntu.com/ubuntu/ubuntu/dists/wily/main/binary-i386/Packages.gz is downloaded; its SHA256 is checked against the SHA256 mentioned in `InRelease`.
- Whenever a `.deb` package downloaded that is mentioned from `Packages`, its SHA256 is checked against the package SHA256 mentioned in `Packages`. 

I expect that a reasonable number of rpi users have logged into their webmail from a Raspberry Pi; it would be nice if they would be safe by default (or shown a warning that they shouldn't expect their rpi to be secure).

lurch, 2015-11-22:
See also #235 (which is similar to, but not the same as, this issue).

In theory the http://downloads.raspberrypi.org/os_list_v2.json *could* be served over https, but as @bennuttall says [here](https://github.com/raspberrypi/noobs/issues/235#issuecomment-91507821) the RPF only have a SSL certificate for the `www` domain. And to remain backwards-compatible with previous NOOBS releases, we'd still have to serve both HTTP and HTTPS versions anyway.
The RPi **does** have an entropy source - a quick search finds [this](http://scruss.com/blog/2013/06/07/well-that-was-unexpected-the-raspberry-pis-hardware-random-number-generator/) old article, but I suspect the setup instructions have probably changed since then. [More info](https://www.google.co.uk/search?q=raspberry+pi+hwrng) (although it's probably not enabled in the current version of NOOBS).

The tarballs themselves (`boot.tar.xz`, `root.tar.xz` etc. which is presumably where any 'virus' would be hidden) are so big that a) they'd be too 'expensive' (in terms of server-load) to serve over HTTPS, b) there's no way to verify their signature before installing them (since we don't have any 'temporary space' into which to download them) - they're simply downloaded and extracted straight to the SD card in a pipeline process, and c) I suspect if I did add any kind of verification procedure (we could theoretically checksum the bytes as they're being downloaded, but we wouldn't know if the tarball was corrupt until after we'd finished writing it to SD card) people would only complain about it slowing down NOOBS.

If you're that concerned that your router might be compromised, then you should use the "offline" version of NOOBS, and verify its SHA1 checksum on https://www.raspberrypi.org/downloads/noobs/

And I have no idea what webmail has to do with NOOBS?!

closed 2015-11-22 by lurch

bennuttall, 2015-11-22:
We now have a wildcard SSL certificate covering the whole raspberrypi,org subdomain namespace.

https://downloads.raspberrypi.org/os_list_v2.json is valid HTTPS.

bgeron, 2015-11-22:
Thank you for the fast replies.

Have you tested that TLS will have a performance impact on the servers? According to [many](https://istlsfastyet.com) [people[(http://stackoverflow.com/questions/149274/http-vs-https-performance), the only real performance impact of TLS is that it adds a bit of latency to set up the connection. My 2012 laptop en/decrypts about 500MB/s of AES; my Raspberry Pi 2 decrypts about 18 MB/s of AES. 

> And I have no idea what webmail has to do with NOOBS?!

I wanted to give an example of (I think) many non-knowledgeable people use a Raspberry Pi to do casual computer stuff that should be secure.

lurch, 2015-11-22:
The other thing I've just remembered, is that "large files" downloaded by NOOBS are served over a CDN (with details [here](https://www.raspberrypi.org/blog/coping-with-christmas-a-guest-post-from-our-web-hosts-at-mythic-beasts/) ) and as that part of the system is out of my hands, I dunno which parts do or don't support HTTPS.