raspberrypi/noobs #291: NOOBS 1.5.0 building

procount, 2015-11-27:
I have some issues building v1.5.0:
1. wifisettingsdialog.h:14:27: fatal error: QDBusObjectPath: No such file or directory
2. VERSION_NUMBER in config.h is still 1.4 not 1.5
3. IIRC, Partclone download URL was incorrect, so I had to do it manually (may just be my problem)

maxnet, 2015-11-27:
>1. wifisettingsdialog.h:14:27: fatal error: QDBusObjectPath: No such file or directory

Double check you have dbus here: https://github.com/raspberrypi/noobs/blob/master/recovery/recovery.pro#L7

procount, 2015-11-27:
Yep, already checked that. And the include file exists, it just can't find be located.
I just did a `git fetch & merge` into the previously released master branch, `make source` to download everything, then `./BUILDME.sh`
(unfortunately, my VMware player has become non-responsive since shutting it down and restarting, so I'm going to have to reinstall it again).

procount, 2015-11-27:
Something's wrong though because QDBusObjectPath exists in ./output/build/qt-4.8.6/include/QtDBus/
and the INCPATH  includes `-I/home/kevin/noobs_dev/buildroot/output/host/usr/arm-buildroot-linux-uclibcgnueabi/sysroot/usr/include/QtDBus`

maxnet, 2015-11-27:
>into the previously released master branch

If you build NOOBS 1.4 previously in the same directory, make sure to force buildroot to rebuild Qt.
E.g. by deleting output/build/qt-4.8.6

NOOBS v1.5 has different Qt build options than NOOBS v1.4
Support for DBus (to talk to wpa_supplicant), OpenSSL  and evdev (for handling absolute touch screen coordinates, as the pc mouse driver only does relative) were not enabled before.

procount, 2015-11-27:
Ah yes.. I've started the long QT rebuild now.
Out of interest, why did you chose to use evdev for the touchscreen instead of tslib, for example?
And why use Dbus/wpa_supplicant, rather than the simpler technique Jumaka used? Is it just as easy to add other wifi dongles? (I'm just trying to learn/understand)

maxnet, 2015-11-27:
>Out of interest, why did you chose to use evdev for the touchscreen instead of tslib, for example?

Qt can read the coordinates directly from /dev/input/event0 (evdev).
No point of using tslib as middle man to do the same, unless you are using a touch screen that requires calibration.

(the touch position is currently some pixels off under NOOBS due to the overscan that shouldn't be enabled when using the touch screen, but once that is fixed it should be spot on)

>And why use Dbus/wpa_supplicant, rather than the simpler technique Jumaka used?

Using dbus instead of calling command line tools has the advantage that it allows two-way communication.
You receive a signal when a new wifi network is detected, or the connection state changes, instead of having to poll for that.

At least that was the theory.
In practice polling whether a DHCP lease was obtained still proved to be a more reliable indicator the connection was working, than the status updates (and that the updates you receive differ depending on which exact wifi authentication type is used, doesn't help either).

> Is it just as easy to add other wifi dongles? (I'm just trying to learn/understand)

Shouldn't be too hard to enable the wifi dongle you have in the kernel configuration.
Not all work reliable enough to be useful to download large operating system images though.

procount, 2015-11-28:
I recloned the repo into a new directory, copied the previously downloaded /dl directory, `make source` to make sure nothing is missing and then `./BUILDME.sh`

When starting to build linux it came up with a missing file:

<pre>drivers/media/radio/Kconfig:22: can't open file "drivers/media/radio/si470x/Kconfig"
make[2]: *** [buildroot_defconfig] Error 1
make[1]: *** [buildroot_defconfig] Error 2</pre>

maxnet, 2015-11-28:
Double check there isn't something more elementary wrong with your build environement, e.g. that you have sufficient disk space.

procount, 2015-11-28:
I have plenty. It's not that, but I think I know what it is now. Will redownload linux source & try again tomorrow. Thanks.

procount, 2015-11-28:
Yay. Fixed my errors. Time to start merging again.

NicoHood, 2016-02-06:
I fixed the other 2 mentioned errors here:
https://github.com/raspberrypi/noobs/pull/305
https://github.com/raspberrypi/noobs/pull/303

Also mentioned here:
https://github.com/raspberrypi/noobs/issues/300

procount, 2016-02-23:
Fixed in 1.6/1.7

closed 2016-02-23 by procount