raspberrypi/noobs #292: NOOBS 1.5.0 Touch Display Issue

chrisgbk, 2015-12-01:
The new LCD display functionality works in the main screen where you select an OS, but when actually installing an OS, the display extends off the screen so the installation progress bar isn't visible.

maxnet, 2015-12-03:
The NOOBS code resizes the progress window to the size of the slide images provided by the operating systems you install.

I suspect you are using NOOBS full, and installed Raspbian, which provides slide images that are larger than the others.

Problem does not show up with NOOBS lite, as a side effect of  http://downloads.raspberrypi.org/raspbian/marketing.tar being broken, and the Raspbian slides being skipped then.
File is gzip compressed, while it should be .tar without compression.

ghollingworth, 2016-02-09:
Fixed in 1.6

closed 2016-02-09 by ghollingworth

NicoHood, 2016-02-09:
@ghollingworth but where is 1.6? #310