raspberrypi/noobs #293: Recovery partition

procount, 2015-12-03:
This is just a general question.
When using a NOOBS formatted SD-card, what is it about the RECOVERY partition (/dev/mmcblk0p1) that prevents it from being automounted in Ubuntu (and probably others) ?
When I insert such a card into my Ubuntu 14.04 PC, all partitions in the extended partition mount, but recovery never does. I always have to mount it manually and it displays as a different icon. If I unmount ANY of the automounted partitions, they ALL unmount, except RECOVERY which I have to unmount manually

lurch, 2015-12-03:
Ahhh, this old chestnut again... ;-)

Duplicate of #262

procount, 2015-12-03:
No, this has nothing to do with the RISC OS.partitioning issue and is not really an issue with NOOBS. (And I am using Ubuntu 14.04 which predates the re-inclusion of that partition type). 
Maybe it's more to do with Ubunutu Udev rules or something. How come Ubuntu will automount the Raspbian FAT Boot partition, but not the Noobs FAT recovery partition?

lurch, 2015-12-03:
Ah, I see, sorry.
Yeah, definitely sounds like an Ubuntu "issue" in that case. Re-opening and tagging as "question".

maxnet, 2015-12-03:
>what is it about the RECOVERY partition (/dev/mmcblk0p1) that prevents it from being automounted in Ubuntu 

The file system label RECOVERY...
If you want it auto mounted, label the file system something else, or remove it from the blacklist at `/lib/udev/rules.d/80-udisks.rules`

```
# recovery partitions (taken from old hal rules)
ENV{ID_FS_TYPE}=="ntfs|vfat", \
  ENV{ID_FS_LABEL}=="RECOVERY|HP_RECOVERY|Recovery Partition|DellUtility|DellRestore|IBM_SERVICE|SERVICEV001|SERVICEV002|SYSTEM RESERVED", \
  ENV{UDISKS_PRESENTATION_HIDE}="1"
```

lurch, 2015-12-03:
Haha, kinda ironic that the only partition that Windows is able to mount, is the only partition that Ubuntu doesn't automatically mount ;-)

procount, 2015-12-04:
So simple, yet not so obvious.
Thank you so much. This has been bugging me for so long. 
[Just thought I'd see how many times I could get the word SO into this reply;) ]

closed 2015-12-04 by procount