raspberrypi/noobs #294: Question - Can I get a debug log out of Noobs like dmesg?

adamst85, 2016-01-04:
Basically I want to see how it is detecting a wireless USB keyboard in order to help fix this issue: https://github.com/raspberrypi/noobs/issues/257

Could someone point me in the right direction and tell me how to get a debug log of sorts from Noobs?

What would be really helpful is being able to see the device tree like in this link:
https://www.raspberrypi.org/documentation/configuration/device-tree.md

ghollingworth, 2016-01-11:
Press Ctrl-Alt-F2 and log in username root password raspberry

dmesg

adamst85, 2016-01-14:
Thanks ghollingworth. That was what I needed :)

closed 2016-02-09 by ghollingworth

NicoHood, 2016-02-09:
FYI: I added this question here:
https://github.com/raspberrypi/noobs/wiki/Developer-Information#can-i-get-a-debug-log-out-of-noobs-like-dmesg

Wanted to ask the same soon ;)