raspberrypi/noobs #295: noobs needs a supported upgrade path

plugwash, 2016-01-12:
The original idea with noobs was that noobs itself would never be updated. The problem with that is that people want to move SD cards between Pis and from time to time hardware changes (either as explicit new models or as component substitutions on existing models) require updates firmware to boo. 

Someone documented that you could upgrade by copying everything except "recovery.cmdline"  However that doesn't seem to work anymore (based on my own test at a jam and a report from mike cook). I guess something changed with how noobs detects installed operating systems but I don't know how noobs internals work.

I was able to get a successful boot on a zero by copying only bootcode.bin and recovery.cmdline from the new noobs zip. However this ends up with a mixed install.

So IMO noobs really needs a supported upgrade path to allow existign SD cards to be used in newer Pi models.

procount, 2016-01-12:
Afaik, noobs 1.5 changed the partitioning scheme. The SETTINGS partition moved from #3 (at the end of the sdcard) to #5 (at the start of extended partition #2) and so all the partitions for the installed O.S.s moved up by one. So the recovery program in noobs 1.5 may not understand an existing 1.4 or earlier partitioned card.
if you have such a card and want to upgrade noobs to work on a Pi Zero, I think upgrading the boot files as you did should be sufficient. Only when you want to reinstall an O.S. would you need to upgrade to the full noobs 1.5 which will erase all the existing partitions anyway.
The new boot files will also allow it to work with the Rpf 7"lcd screen.
However, if you want to use the touchscreen interface or use one of the rpf wifi dongles then you need the full 1.5 noobs which then won't understand the old partition layout.

lurch, 2016-01-13:
For various reasons, NOOBS is explicitly non-updateable. The only way to get a newer version of NOOBS is to format the entire SD card and start again from scratch with the latest NOOBS zipfile.

**However**, if all you're wanting to do is boot an older version of NOOBS in a newer-released Pi model, then in theory (I haven't tested this myself yet) all you need to update are the boot files `bootcode.bin` and `recovery.elf` (no other files should be modified). These can either be copied from the latest NOOBS zipfile, or you can get them from https://github.com/raspberrypi/firmware/tree/master/boot (`recovery.elf` used by NOOBS is simply a renamed `start_cd.elf` from the firmware repo).

closed 2016-01-13 by lurch