raspberrypi/noobs #296: Partition noobs or delete partition

matteos1, 2016-01-13:
i've a sd with 32 gb with installed with noobs , raspbian and osmc.
the memory on partition with osmc is low and i would like to delete the partion with rapbian or decrement the partition with raspbian to increase the partition with osmc.
is it possible to dimension the partition??
thank a lot

lurch, 2016-01-13:
The only way for NOOBS to do this is to press SHIFT at power-on, and then choose to install just OSMC without also installing Raspbian (which will then allocate all available space to OSMC). However this will involve deleting your existing OSMC installation first.

NOOBS has no support for resizing existing installations, although (if you write down which OSes are installed onto which partitions) you *might* be able to do that using an SD card-reader and something like GParted on a standalone Linux PC. But this is obviously officially unsupported and entirely at your own risk ;-)

closed 2016-01-13 by lurch

matteos1, 2016-01-13:
why noobs haven't the option to reduce or delete partition? could be very usefull

lurch, 2016-01-13:
Because NOOBS isn't a full-fledged partition editor! ;-)

Imagine for example if Raspbian was on partitions 5 and 6 and OSMC was on partitions 7 and 8. If you then deleted Raspbian's partitions, you'd need to A) move OSMC's partitions to the start of the available free space, B) enlarge (what is now) partition 6, C) resize the filesystem on partition 6 to use all the available space, D) edit the relevant files in OSMC to tell it it's now on partitions 5 and 6 instead of 7 and 8.

All well outside the scope of "New Out Of Box Software" (apart from step D), and jobs for a full partition editor program.

And of course it gets even more complex if you have more than 2 OSes installed in NOOBS...

matteos1, 2016-01-13:
and if i make a backup of sd and restore on sd more larger?
the dimension of free space change?

lurch, 2016-01-13:
Due to the way the [MBR partitioning](https://en.wikipedia.org/wiki/Master_boot_record#Disk_partitioning) works (all of the NOOBS 'OS' partitions are logical partitons inside an extended partition) you'd still need to use an external partition-editing tool to resize your existing partitions.

There's only **one** time when NOOBS resizes a partition, and that's the very first time you boot it on a fresh SD card, when it then resizes the first FAT32 partition (as small as possible) to make space for all the 'OS' partitions that get created when you use NOOBS to install other OSes.