raspberrypi/noobs #298: documentation of NOOBS outdated with NOOBS 1.5.x new partition layout

beta-tester, 2016-01-17:
the partition layout of the documentation 
e.g. on https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained
the partitions are described like
primary partitions
1'st (/dev/mmcblk0p1) = NOOBS boot
2'nd (/dev/mmcblk0p2)  = extended partition
3'rd (/dev/mmcblk0p3)  = NOOBS settings
4'th  (/dev/mmcblk0p4) unused

extended partitions
2.1'st (/dev/mmcblk0p5)  = logical partition with first installed OS boot partition
2.2'nd (/dev/mmcblk0p6)  = logical partition with first installed OS root partition

2.3'rd (/dev/mmcblk0p7)  = logical partition with second installed OS boot partition
2.4'th (/dev/mmcblk0p8)  = logical partition with second installed OS root partition
...
2.n'th (/dev/mmcblk0pX) = optional 512MB user data partition if installed

but with NOOBS 1.5.x the described layout is obsoleted.
the NOOBS 1.5.x partition layout is like
primary partitions
1'st (/dev/mmcblk0p1) = NOOBS boot
2'nd (/dev/mmcblk0p2) = extenden partition
3'rd (/dev/mmcblk0p3)  unused
4'th  (/dev/mmcblk0p4) unused

extendend partitions
2.1'st (/dev/mmcblk0p5) = NOOBS settings

2.2'nd (/dev/mmcblk0p6) = logical partition with first installed OS boot partition
2.3'rd (/dev/mmcblk0p7) = logical partition with first installed OS root partition

2.4'th (/dev/mmcblk0p8) = logical partition with second installed OS boot partition
2.5'th (/dev/mmcblk0p9) = logical partition with second installed OS root partition
...
2.n'th (/dev/mmcblk0pX) = optional 512MB user data partition if installed

so the NOOBS settings partition is moved from 3'rd primary partition to the first logical partition with shifting all the following logical partitions one up in their order.
but it is not in the documentation that describes the new layout either that the layput of NOOBS <=1.4.x differs to NOOBS 1.5

procount, 2016-01-22:
Also, new features need documenting, (e.g. the use of wpa_supplicant.conf)

XECDesign, 2017-11-03:
Procount has already updated partition layout documentation

closed 2017-11-03 by XECDesign