raspberrypi/noobs #302: integration of exFAT-Support

XD1990, 2016-02-03:
Today i tried again without any success to install noobs on a 64gb SD-Card. 
I know that ExFAT is actually not support by Noobs.

It would make sense to integrate it sooner than later.
So why not earlier.

Greetings

procount, 2016-02-03:
This has nothing to do with NOOBS.
The Pi is started by the GPU which only knows how to read a FAT32 partition to load the bootfiles (and I believe this is hardcoded into the ROM of the GPU so it is unlikely to change - please correct me if this is an incorrect assumption).
So you will find that every distro must have its boot files stored on a FAT 32 partition: Raspbian, Arch, OSMC, Ubuntu etc, even NOOBS if you consider that to be a small distro.

XD1990, 2016-02-03:
yeah but the other disto in my example raspbian works well with the standard dd command.

procount, 2016-02-03:
I'm not sure I understand your comment.
Do you mean that you can write a Raspbian.img file to an SDcard using DD from your linux box?
If so, then you are no longer a "newb" and probably not the main target audience for NOOBS which was primarily designed for "newbs" who probably have a Windows machine and want to easily write an SD card to get started with the Pi and dip their toe into the Linux world. For some of these people, using Win32DiskImager was considered to be too difficult. (This is in addition to all the other reasons for designing NOOBS: easy recovery when a distro gets messed up, enabling multi-boot selection etc)
There is nothing to stop you creating a NOOBS.IMG file which you can use to install NOOBS onto any SD card using the dd command from your linux box. This way you wouldn't have to ensure your card was FAT32 first (but make sure the IMG file is created from a FAT32 partition in the first place otherwise it will not boot on the Pi)

closed 2016-02-03 by XD1990

lurch, 2016-02-03:
@procount is correct about the first-stage bootloader being hardcoded into the GPU's ROM :-)

For more info on NOOBS and exFAT, see https://www.raspberrypi.org/documentation/installation/sdxc_formatting.md

The reason that Raspbian 'works' when dd-ed onto an 'exFAT' SD card, is that dd is at such a low-level it operates 'below' the filesystem (and overwrites the MBR), and so once you've finished writing Raspbian to your SDcard using dd, you'll find that it no longer has an exFAT partition, but instead has a small FAT32 partition at the beginning.