raspberrypi/noobs #304: Unable to mount root fs

golddave, 2016-02-05:
I ran an 'apt-get update' and 'apt-get upgrade' on my PI and it no longer boots. I'm getting a "Unable to mount root fs" error. I booted into NOOBS (My card has 1.3.9 installed) to check the cmdline.txt (as I've seen other people with the same issue being asked to do) and it says the following:
dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tt1 root=/dev/mmcblk0p6 rootfstype=ext4 elevator= deadline rootwait

Any help would be greatly appreciated.

Thanks,
Dave

lurch, 2016-02-07:
I'm away from home this week, but will investigate this next week.
I assume that you're talking about Raspbian installed inside NOOBS v1.3.9, and that it booted fine before this recent OS-update?

Do you have a Linux PC with an SD card-reader you can insert this card into, to check what data is actually on the other partitions?

golddave, 2016-02-07:
Your assumption is correct (Raspbian inside NOOBS v1.3.9 booted fine before update).

I have a Mac with an SD card slot. Using it I got the following info from the BUILD-DATA file on the recovery partition:
Build-date: 2014-07-08
NOOBS Version: v1.3.9
NOOBS Git HEAD @ 5e64133d93a3ef8870dcf19caaf9c4a44fd1b979
rpi-userland Git HEAD @ 5fd0157bcb1f28baba1316f5ac5439d90ed31774 refs/heads/master
rpi-firmware Git HEAD @ 462f3e3f476f7b6200004040c09e992f3c551f38 refs/heads/master

Also, from the text console in NOOBS I ran an fdisk -l /dev/mmcblk0 which gave me the output in the attached image:
[image]

Also, this is the actual screen output I'm seeing when I boot the Pi:
[image]

maxnet, 2016-02-07:
Try ``fsck.ext4 -yf /dev/mmcblk0p6``

golddave, 2016-02-08:
Can I run that from the text console in NOOBS? Or do I need another Pi or Linux system to run it? (Mac says "command not found: fsck.ext4".)

maxnet, 2016-02-08:
NOOBS console

golddave, 2016-02-08:
It took a few minutes to run. When it was done I rebooted the Pi and it gave me a different error. Now I get the following:
VFS: Mounted root (ext4 filesystem) readonly on device 179:6.
devtmpfs: error mounting -2
Freeing unused kernel memory 352K (c07c6000 - c081e000)
Kernel panic - not syncing: No working init found. Try passing init= option to kernel. See Linux cocumentation/init.txt for guidance.

This is followed by a few lines with the headings unwind_backtrace, show_stack, dump_stack, panic and kernel init. Last line says:
random: nonblocking pool is initialized.

Then it stops. This is a screenshot (but it's not so good.
[image]

maxnet, 2016-02-08:
You can try if the files are still accessible with the (older) kernel version used by NOOBS, from the NOOBS console.

```
mount /dev/mmcblk0p6 /mnt2
ls -l /mnt2
*check if there are any folders*
dmesg
*check for any errors*
```

But if not, changes are high your file system is corrupt beyond usable.

golddave, 2016-02-08:
The ls -l shows a directory called "lost+found" as follows:
[image]

Also,  the dmesg results in:
EXT4-fs (mmcblk0p6): couldn't mount as ext3 due to feature incompatibilities
EXT4-fs (mmcblk0p6): couldn't mount as ext2 due to feature incompatibilities
EXT4-fs (mmcblk0p6): mounted filesystem with ordered data mode. Opts: (null)
Here is the screenshot:
[image]

golddave, 2016-02-09:
Booting again this morning I got the same as I got 2 updates ago. Screenshot:
[image]

maxnet, 2016-02-09:
>The ls -l shows a directory called "lost+found"

If there are no folders in the root other than that left, I would consider the file system corrupt beyond recoverable, and reinstall.

goldsteindavid, 2016-02-11:
Do you think the SD card may be damaged? Should I get a new one?

golddave, 2016-02-11:
Do you think the SD card may be damaged from this? Should I get a new one?

closed 2016-02-16 by golddave