raspberrypi/noobs #307: Add Ubuntu Mate Builds to Noobs

NicoHood, 2016-02-06:
It would be cool to support Ubuntu Mate in Noobs, since its also listed on the official website and runs very stable.

In this thread I'd like to discuss the manual addition of an ubuntu mate build and in the end it would be cool to officially add it to NOOBS.

**Note that this is not going to work at the moment. I am just documenting my steps so people can have a look at it and point me maybe into the right direction and help each other.**

The steps I made so far:
```bash
# Check partition offset with fdisk (p, q)
fdisk ubuntu-mate-15.10.1-desktop-armhf-raspberry-pi-2.img

# Create temporary directories for mounting the raw image
mkdir /tmp/mnt
mkdir /tmp/mnt/boot
mkdir /tmp/mnt/root

# Mount the partitions with the offset noted with fdisk. You have to multiply them by 512.
sudo mount -o loop,offset=1048576 ubuntu-mate-15.10.1-desktop-armhf-raspberry-pi-2.img /tmp/mnt/boot
sudo mount -o loop,offset=68157440 ubuntu-mate-15.10.1-desktop-armhf-raspberry-pi-2.img /tmp/mnt/root

# Pack the files. Root has to be packed with root user.
cd /tmp/mnt/boot
tar -Jcf /tmp/boot.tar.xz .
cd /tmp/mnt/root
sudo tar -Jcf /tmp/root.tar.xz .

# Now make the file available to normal users again
sudo chown username /tmp/root.tar.xz 
sudo chmod 755 /tmp/root.tar.xz 

# Get the size of the extracted tar archive we created above.
cd /tmp/mnt/
sudo du -s root/
sudo du -s boot/
```

Now create a folder inside `os` named `Ubuntu`. Place the two created tar.xz files in there.
You also need to create a few other files:

**TODO partition_setup.sh was not checked if it works and edits the correct files**

partition_setup.sh
```
#!/bin/sh

set -ex

if [ -z "$part1" ] || [ -z "$part2" ]; then
  printf "Error: missing environment variable part1 or part2\n" 1>&2
  exit 1
fi

mkdir -p /tmp/1 /tmp/2

mount "$part1" /tmp/1
mount "$part2" /tmp/2

sed /tmp/1/cmdline.txt -i -e "s|root=/dev/[^ ]*|root=${part2}|"
sed /tmp/2/etc/fstab -i -e "s|^.* / |${part2}  / |"
sed /tmp/2/etc/fstab -i -e "s|^.* /boot |${part1}  /boot |"

umount /tmp/1
umount /tmp/2
```

**The sizes of the files are not correct. How is the correct schema there? what about `want_maximised` setting?**

partitions.json
```json
{
  "partitions": [
    {
      "label": "boot",
      "filesystem_type": "FAT",
      "partition_size_nominal": 63,
      "want_maximised": false,
      "uncompressed_tarball_size": 21
    },
    {
      "label": "root",
      "filesystem_type": "ext4",
      "partition_size_nominal": 3482,
      "want_maximised": true,
      "mkfs_options": "-O ^huge_file",
      "uncompressed_tarball_size": 3082
    }
  ]
}
```

**It would be cool to add a note that this is only supported with the pi2 or so. how to do that? Anything else I missed?**

os.json
```json
{
  "name": "Ubuntu Mate",
  "version": "wily",
  "release_date": "2015-12-21",
  "kernel": "4.1",
  "description": "Ubuntu MATE for the Raspberry Pi 2",
  "url": "https://ubuntu-mate.org/raspberry-pi/",
}
```

The Icon for Ubuntu is also missing, but not important.

Could someone please check the steps I made and help me correct the files above?

lurch, 2016-02-07:
Yup, you're heading along the right lines, well done :-)

### Questions answered

> Is it really important to compress the images that high? Cant I just do it with tar -J option? It takes forever on my i7, on the pi it might be even more for unpacking?

Primary reasons for such high compression are so that it a) takes up less space on the first partition of the SD card (for offline NOOBS) leaving more space on the rest of the card for the installed OSes b) takes less time to download the tarballs if you're using online NOOBS. xz is quite quick to decompress, and although I've never actually benchmarked it, I suspect the SD card-writing speed is probably more of a bottleneck? IIRC the xz-decompression is always the same speed, regardless of the compression ratio.
For testing purposes (where you don't care if the files are bigger than they need to be), you're obviously free to use a much lower compression ratio.

> Are my partition sizes calculated correct? Please add a concrete method in the readme to get those values.

I obviously don't have time to double-check your figures, but instead of doing `du` on the mounted directories, you could just do `ls -l` on the tar files before you compress them :)

> It would be cool to add a note that this is only supported with the pi2 or so. how to do that in the os.json?

Yup, that's what the `supported_hex_revisions` is for - look at the differences between e.g. https://downloads.raspberrypi.org/osmc_pi1/os.json (only compatible with Pi1) and https://downloads.raspberrypi.org/osmc_pi2/os.json (only compatible with Pi2) and 
https://downloads.raspberrypi.org/raspbian/os.json  (compatible with all Pis)

I did start working on code so that only Pi2-compatible OSes would get listed on a Pi2, and only Pi1-compatible OSes would get listed on a Pi1, but unfortunately I haven't got round to finishing testing and merging it yet.

> Anything else I missed?

Yup, as well as the icon, there's also the installation slides, which are in the `slides_vga` folder in offline NOOBS (i.e. on the SD card), and `marketing.tar` for online NOOBS (i.e on https://downloads.raspberrypi.org/ ).

### Other points

* I think it'd be better if your `os` subdirectory was named `Ubuntu_Mate` rather than just `Ubuntu`. I can't remember for sure, but I think the name of the `os` subdirectory *might* need to match the name in `os.json` (with spaces replaced with underscores, like in https://downloads.raspberrypi.org/data_partition/os.json

* The bit where you use `fdisk` and `mount` to loopback-mount your disk-image with offsets is made much easier by using `kpartx` (can't remember the details off the top of my head, but I think it's something like `kpartx -av disk.img` and then you can directly `mount` the `/dev/mapper/loop0p1` etc. that get created with no need for offsets)

* When you get a bootable setup working, you might want to do a test-install with `want_maximised` set to false for both partitions, so you can be sure that there's enough space for the OS to boot fully, including any temporary files (e.g. swap) that may get created during the first bootup.
If the user is squeezing several OSes onto their SD card, then (if you're unlucky) your root partition _may_ get created at `partition_size_nominal`, even with `want_maximised` set to true.

You're right that this should all be much better documented, but unfortunately documentation always tends to slide to the bottom of my to-do list!

NicoHood, 2016-02-07:
Thanks for answering my questions in detail!

> I did start working on code so that only Pi2-compatible OSes would get listed on a Pi2, and only Pi1-compatible OSes would get listed on a Pi1, but unfortunately I haven't got round to finishing testing and merging it yet.

It would be cool to not hide them while installing, but while booting. Then you can use the sd card on a pi zero and on a Pi2. I usually do this very often and switch to different pis. Especially with raspbian. This should not keep me from installing ubuntu mate for pi2 only if i set it up on pi 1. A not would be okay, but it should not be hided. In the boot menu later it can be hided, that makes sense.

#### Another Question:
I tried to mount the sd card with my PC. I get (after deleting risc os):
```
sdc                                                    8:32   1  29,5G  0 disk  
├─sdc1                                                 8:33   1   3,2G  0 part  
├─sdc2                                                 8:34   1     1K  0 part  
└─sdc5                                                 8:37   1    32M  0 part
```
I can mount the setting partition via file manager, but not the sdc1. I can mount it manually though. Any reason why the file manager hides it? I'd like to change this.

The wiki is available for editing. I guess I am supposed to do that?

procount, 2016-02-07:
sdc1 is not auto-mounting because its label name = RECOVERY. Change it to something else and it should automount.

NicoHood, 2016-02-07:
Thank you very much. I will try to add as much information as I can to the wiki:
https://github.com/raspberrypi/noobs/wiki/Adding-custom-OS-version-examples

I also contacted Martin from Ubuntu Mate Team to have a look at this.
Having official Ubuntu Mate builds is not a big deal and would really make Noobs a lot worth to use.

@procount I will also test your Arch fork now. I am just not sure how to extract the single .tar.gz file into two partitions properly. But I got a workaround and will report the results later in the other thread =)

If you also have some useful information to add to the wiki, why not do it ;)

lurch, 2016-02-07:
> Thanks for answering my questions in detail!

You're welcome!

> It would be cool to not hide them while installing, but while booting. Then you can use the sd card on a pi zero and on a Pi2.

My plan was to hide the 'incompatible' OSes both for installation and booting, since IMHO that's what most people will expect. But for people like you who want to use the same NOOBS SD-card in both Pi1 and Pi2, I was planning to have a `recovery.cmdline` parameter you could add, to show **all** OSes in the install-list, even those that are not compatible with the Pi the SDcard is currently in.

> The wiki is available for editing. I guess I am supposed to do that?

Yeah, feel free to edit if you have useful content to add... and don't forget you can also submit PRs to https://github.com/raspberrypi/documentation if you have non-technical NOOBS info that you think would be helpful to everyday NOOBS users https://www.raspberrypi.org/documentation/installation/noobs.md

> What is feature level?

`feature_level` is what `supported_hex_revisions` *used* to be called in older versions of NOOBS, and is a 64-bit enum / bitstring. However with the new versioning scheme that was introduced with the Pi2, a 64-bit value was no longer big enough, and so we switched to a comma-separated string instead. `feature_level` doesn't need to be in the `os.json` file stored on the SD-card (as it's not used by current versions of NOOBS if `supported_hex_revisions` is present) but it *does* need to be in the `os.json` files stored on https://downloads.raspberrypi.org/ so that any old versions of NOOBS Lite (from before `supported_hex_revisions` was added) don't suddenly stop working.
`feature_level` can be automatically calculated from `supported_hex_revisions` using:
```python
def supported_hex_revisions_to_feature_level(hex_revisions):
    feature_level = 0
    for hex_revision in hex_revisions.strip(' ').split(','):
        int_revision = int(hex_revision, 16)
        if int_revision < 64:
            feature_level |= 1<<int_revision
    return feature_level
```

Note also that for people 'allergic' to hexadecimal numbers (LOL!) there's also `supported_revisions`, which is identical to `supported_hex_revisions` except the numbers are in decimal (base 10) rather than hexadecimal (base 16). But `supported_hex_revisions` is preferred over `supported_revisions` where possible.

> ls -l does not work for me. What did I do wrong?

Re-read what I wrote more carefully ;-)  "you could just do ls -l **on the tar files** before you compress them"

> Any reason why the file manager hides it? I'd like to change this.

There's a bit more elaboration of what @procount is talking about with 'RECOVERY' at https://github.com/raspberrypi/noobs/issues/293#issuecomment-161753230 (it's another Ubuntu-specific thing rather than a NOOBS thing). But as you've seen it's also easy to workaround by manually using the mount command, or IIRC you can also mount your `sdc1` using the "Disks" application in Ubuntu.

lurch, 2016-02-07:
P.S. You don't need `slice_exclude` - that was only used many years ago when the original version of https://www.raspberrypi.org/downloads/ was auto-generated from the information found in the `os.json` files - which is no longer the case.

procount, 2016-02-07:
@NicoHood - my Arch fork does not need the tar.gz file to be extracted or split. It uses the one provided by the maintainers with a bit of partition_setup.sh changes. I am in the process of pushing all of it to the internet so if you can wait a bit I'll have done it for you, or study my source & guess what you have to do ;-)

NicoHood, 2016-02-07:
@lurch can you please add the answer for "What is feature level?" to the wiki page? If you do it, you can just copy the markdown format. That'd be nice. I suggest somewhere above this entry:
https://github.com/raspberrypi/noobs/wiki/Adding-custom-OS-version-examples#other-resources

I also collect other information that I found useful when chatting with you guys here:
https://github.com/raspberrypi/noobs/wiki/Developer-Information

NicoHood, 2016-02-29:
@lurch last question about the wiki is still valid, also I asked you some things via mail and never got one back.

What about adding Ubuntu Mate to Noobs by default, without a custom build?

lurch, 2016-02-29:
Apologies for the lack of responses (both here and over email), but I've been very busy recently (working on non-Pi projects), and I'll get back to looking at NOOBS when I can (probably not until next week at the earliest).

ghollingworth, 2016-03-01:
First Ubuntu Mate would need to build a compatible build version (i.e. provide the correct files for a NOOBS release) and then we could look at adding the pointers to the os_list_v3.json to the build.

It will be Ubuntu Mate's responsibility for providing the build and supporting users....

MartiniGM, 2017-02-26:
When I use the method describes in the Wiki Ubuntu Mate 16.04.2 fails when the Installation/Setup Wizard starts. I have better experience with finishing the installation of UbuntuMate the regular way (write image as single OS, run once and finish setup) and then making the root and boot file pretty much the same way you describe in your wiki, but from a working, post-setup UbuntuMate installation instead of making it from the install image.

XECDesign, 2017-11-03:
Answered

closed 2017-11-03 by XECDesign