raspberrypi/noobs #308: Noobs Lite 1.5 erasing all file system without ask when first runing

ZKxZK, 2016-02-06:
Please programing in noobs, when first runs Noobs do not change file system. After  begin instalation Noobs questing whether the modified MBR and all partititons. 

Example when rewritten all the files of noobs, so when you start raspberry will be unnecessary erase all data on the SD card. 

Much better would be if noobs overwriting asked after begin instalation. Example questions may, for example on:
1) The use of existing disk-green?
2) Totally erasing all data and then repartition your disk (yes/no)?

Automatically deleted all dates on SD card without asking agreement is a very dangers function. I came so completely careless by all dates before begins to run installation!

Thanx very much.

lurch, 2016-02-07:
I'm having a hard time understanding your question, but I'm guessing that you had a SD card with an older version of NOOBS on it, and then you copied NOOBS Lite v1.5 onto the first partition, and then when you next booted the SD card you were surprised when NOOBS then 'wiped out' all your old partitions?
(the partitions are not actually deleted until you install an OS using NOOBS, but as NOOBS overwrites the MBR then I guess you no longer know where your old partitions start and stop)

I'm afraid NOOBS simply isn't "upgradeable" in this way (never has been, never will be).
If you want to use a newer version of NOOBS, without wiping out your old partitions, then you'll need to leave the old version of NOOBS on your old SD card, and install the new version of NOOBS onto a new (empty) SD card.

closed 2016-02-07 by lurch

ZKxZK, 2016-02-07:
Thanks for answer,
in that case, when you run a showing on display this query alert: "Next step is modifying SD card, dangers, noobs erase all partitions and dates!" Continue this step yes / no"? 

The query should be no problem to create. In case of a don't disagreement, stopping program noobs.

Seems to me Without warning everything  to delete is the bad, by ginners  learn this the device.

lurch, 2016-02-07:
The [NOOBS setup instructions](https://www.raspberrypi.org/help/noobs-setup/) explicitly tell you to format the SD card first, so showing a "Do you want to erase all partitions?" popup during NOOBS first boot would simply be an annoyance 99.9% of the time.

ZKxZK, 2016-02-07:
I agree with you, once to a problem occurs with the host system (for example, will not boot OS because of a damaged or defective set of file system for example due to a power outage, you want to perform a manual upgrade noobs bootloader, etc ...) This message is really extra step. Anyway good experience with noobs, when carelessness or failures HW or SW automatically completely spontaneously permanently erases everything. It is a pity.

lurch, 2016-02-07:
NOOBS never writes to the first partition on the card (containing the boot files, and the files for NOOBS itself), so it's impossible for a NOOBS system to become unbootable, even in the event of a power outage.
As mentioned before, NOOBS isn't designed to be upgradeable, it's only a "start from a clean slate" system (especially with the recent changes from NOOBS v1.4.x to NOOBS v1.5.x where the settings partition moved to a different number). Just out of curiosity, what exactly were you hoping to achieve with your "manual NOOBS upgrade"? I always pay careful attention to backwards-compatibility, so AFAIK even older versions of NOOBS Lite should still be able to install the latest OS files.

ZKxZK, 2016-02-07:
Thank you for question, description of the situation.

1) New installation Pi 2 Noobs over 1.4.x Lite, function OK and install Rasbian 7
2) Upgrade to Rasbian 8 Jessie
3) After months of nonstop operation one month, OS halt with kernel panic
4) Restart Pi2 and after 1 second starting Rasbian 8 kernel panic
5) Run Noobs Lite 1.4.x, but I could not find opportunity to start in safe mode (similar to the grub2)
6) Deleting all files Noobs 1.4 Lite and embedded files Noobs Lite 1.5.x
7) Start Pi2 and launch Noobs 1.5 Lite
8) Noobs automaticali and quickly remove and erase all existing file systems and all dates and created new and empty partitions for new installations (Noobs begin installation notwithstanding on current installation). I did not wish to install, I searched features for repair options in new version Noobs 1.5
9) Attempt recovery dates on another linux, testdisk and came out gparted ...resulc is bad
10) I written query on this forum with the hope to protect other future users

Thank you very much for accommodating communication, I appreciate it.

procount, 2016-02-08:
When changing/upgrading NOOBS in this (non-recommended) way, it is important to check the recovery.cmdline file. A new version of this file has `runinstaller` in it which causes the SD card to be reformatted and re-initialised. Removing this command word would have avoided the deletion. But it may still not have worked in this instance due to the change in partition layout between v1.4 and v1.5.

ZKxZK, 2016-02-08:
Yes, it would be enough to alert and stop the program before erasing. I deem especially important not use the automatic deletion of all data without warning. I made a nonsense, and he get me cruel punishment, without warning, I immediately lost all the data. I did not read none information, unfortunately, that do not upgrade noobs.