raspberrypi/noobs #312: sort order of os_list.json

beta-tester, 2016-02-09:
hi, could somebody sort the entries of the available OS'es in a logical order to keep Pi1 and Pi2 versions of OS'es together.
http://downloads.raspberrypi.org/os_list_v3.json
for example OSMC_Pi1 and  OSMC_Pi1 are in the list, but there is in between there are severyl other OS entries listed.
so it is possible, that somebody will see  OSMC_Pi1, but do not realize, that there is a  OSMC_Pi2 version available further down as well, because those two entries are not close together.

options i see here:
option 1: edit the  os_listXXX.json file by hand to bring the entries in a good order. quick and easy.
option 2: implement a alphabetical sort order functionality to NOOBS. code changes necessary to do that. no changes to os_listXXX.json file.
option 3: implement a ranking + alphabetical sort order functionality to NOOBS. code changes necessary + structural changes of os_listXXX.json necessary for adding a ranking order attribute.

procount, 2016-03-11:
NOOBS v1.7 will now only show the OS'es that are applicable to your hardware model. It can be forced to show all available OS'es by adding the `showall` cmdline option.
Does this solve your problem enough, or do you think sorting is still required on top of this?

beta-tester, 2016-03-12:
that solution sounds good enough - then there is no possibility that a newbie can install an incompatible OS to his/her RPi.
thanks

sorting by ranking and alphabetical on top would be nice, but way too much.
the person who editing the os_list files, at least, should take more care about, that the entries appear in a nice logical order, for each OS. for each RPi version the same!
to kill any chance for any confusion.

closed 2016-03-12 by beta-tester

beta-tester, 2016-03-12:
i just took a look to the http://downloads.raspberrypi.org/os_list_v3.json file and noticed the "supported_models" entries... when "Pi 3" is not in the list, will it mean it will not be shown in NOOBS as available?
because for some OSes i see no "Pi 3" but i know that a version of the OS is available that supports RPi 3 already,
but i guess i am hitting an other problem with that - that they didn't provided a NOOBS compatible image just in time yet...

procount, 2016-03-12:
Thank @maxnet not me...I'm just pointing it out :)
The supported_models replaces the other techniques and works as you describe.