raspberrypi/noobs #316: Question - network automated install, howto ?

fthiebolt, 2016-02-26:
Hello,

Noobs works like a charm for OS install from SDCard (i.e you already put an OS and it will install from SDCard) ... but on my side i have a bunch of RPis connected to my university network and I want a kind of PXE network automated installation.
Right now, i already modded recovery/config.h to change link to my own server but i'd like to have in recovery.cmdline some kind of additional kernel parameter like 'silentinstall distrib=myCustomRaspbian' and then, once network is up and os_json parsed, it will start to download and install the distribution.
At this time, my RPis can connect to my server and start installation from network but i want it fully automated!

I guess this is just a matter of launching an event once DownloadLists() is over ...

Thanks
François

lurch, 2016-02-26:
Yeah, I think @procount and @meltwater2000 have asked for a similar feature before too, it just hasn't been written yet.

Have you looked at http://pinet.org.uk/ to see if that does anything similar to what you're looking for?

procount, 2016-02-26:
If you have a look at my procount/noobs repository in the new_alt_image_150 branch, you can see a more generic approach to allow downloading from a different server without having to hardcode it in the source. This was for v1.5, but I have migrated it to v1.7 now, so expect a new_alt_image_170 branch to appear on there soon (or I'm sure you can adapt the v1.5 version as required)

There is a silentinstall cmdline option already, but I think it only works from the SD card if there is a single distro stored there.

Yes, @meltwater2000 asked for this a while back. It has been on my todo list for a while, but I've not got around to it yet.

fthiebolt, 2016-02-26:
@procount yes I already saw your PiNN, but if you activate silentinstall, network is not setup ... anyway i will have to mod the source (what i'm currently doing!) to find the right place once osèjson files are downloaded and to emit an event that will select thr proper image to start its install :)

procount, 2016-02-26:
@fthiebolt - For a generic solution, perhaps you could also consider installing distros from elsewhere (e.g. SDcard) as well and cover the cases where the network is not available....? This would mean not relying on DownloadLists() executing at all, but allowing it to occur if possible.

lurch, 2016-02-26:
...or perhaps you could leave the existing `silentinstall` as-is (only installing from SDcard) and then add a new `silentnetworkinstall` option which installs from the network (and fails if networking isn't available) ?

fthiebolt, 2016-02-26:
@lurch you're right, `silentnetworkinstall` is the right option. Anyway, what i need is a way to fully reinstall a whole OS from network (with noobs) for setting a whole new version of my custom Raspbian (because i have +/- 30 RPis hidden in ceiling and it is a pain to get access to them :|)

I'm currently trying to add some code to check if ui->list->count != 0 at the end of downloadListComplete() ... we'll see

fthiebolt, 2016-02-26:
it works!
Ok, now i should add the `silentnetworkinstall` option and test that once my custom Raspbian has booted, i can find a way to reactivate nOObs on subsequent reboot to restart a whole new installation of my latest Raspbian version.

procount, 2016-02-26:
Well done!
To force NOOBS to run on a reboot in order to update, maybe look at adding `forcetrigger` to recovery.cmdline, But you need to delete it afterwards, otherwise it will never boot into Raspbian on its own anymore.

fthiebolt, 2016-02-26:
```
void MainWindow::downloadListComplete()
{
    QNetworkReply *reply = qobject_cast<QNetworkReply *>(sender());
    int httpstatuscode = reply->attribute(QNetworkRequest::HttpStatusCodeAttribute).toInt();

    /* Set our clock to server time if we currently have a date before 2015 */
    QByteArray dateStr = reply->rawHeader("Date");
    if (!dateStr.isEmpty() && QDate::currentDate().year() < 2015)
    {
        // Qt 4 does not have a standard function for parsing the Date header, but it does
        // have one for parsing a Last-Modified header that uses the same date/time format, so just use that
        QNetworkRequest dummyReq;
        dummyReq.setRawHeader("Last-Modified", dateStr);
        QDateTime parsedDate = dummyReq.header(dummyReq.LastModifiedHeader).toDateTime();

        struct timeval tv;
        tv.tv_sec = parsedDate.toTime_t();
        tv.tv_usec = 0;
        settimeofday(&tv, NULL);
    }

    if (reply->error() != reply->NoError || httpstatuscode < 200 || httpstatuscode > 399)
    {
        if (_qpd)
            _qpd->hide();
        QMessageBox::critical(this, tr("Download error"), tr("Error downloading distribution list from Internet"), QMessageBox::Close);
    }
    else
    {
        processJson(Json::parse( reply->readAll() ));
    }

    reply->deleteLater();
//
// MOD START HERE
        if (ui->list->count() != 0)
        {
            ui->list->setCurrentRow(0);

            if (_allowSilent)
            {
                // No OS installed, perform silent installation
                qDebug() << "Performing silent installation";
                _silent = true;
                ui->list->item(0)->setCheckState(Qt::Checked);
                on_actionWrite_image_to_disk_triggered();
            }
        }
```

lurch, 2016-02-26:
...or (and IIRC this is something I looked at before once) you can force a re-install **without** modifying the boot (mmcblk0p1) partition - as per the NOOBS prime directive - by trashing the (extended) partition-table on mmcblk0p2 by running e.g. `sudo dd if=/dev/zero of=/dev/mmcblk0p2 bs=1M count=4` and then rebooting. https://github.com/raspberrypi/noobs/blob/master/recovery/mainwindow.cpp#L238 will then find that `FAT_PARTITION_OF_IMAGE` no longer exists, and so run another automatic silentinstall.

But obviously you'll want to check that this still works with the new partition-layout in the latest version of NOOBS!

fthiebolt, 2016-02-26:
@lurch I think this solution is OK when you know that you need to force a reinstall
what i'm thinking about is a way to stop getting stuck with RPi unreachable (because of misconfiguration for example) so if the whole boot process didn't reach my rc.local script for some time (RPi watchdog), then i just reboot because there is still the `forcetrigger` option in recovery.cmdline.
More generally, I will have this `forcetrigger` option always set in the early boot of Raspbian ... and it will get disabled when i reach target.reboot (systemd) where my script will remove it for regular reboot. I'm pretty sure there are cases I didn't though of so we'll see in the next few months!
Anyway, thanks for all the tips ;)

lurch, 2016-02-26:
> (because i have +/- 30 RPis hidden in ceiling and it is a pain to get access to them :|)

https://twitter.com/Raspberry_Pi/status/693148645038952448

fthiebolt, 2016-03-08:
Hi Andrew,
Well, I tried your ( a bit hard) method to force a network reinstall by deleting the extended parition ... and it does not work displaying a for ever "waiting for settings partition".
I even tried to add `runinstall` at recovery.cmdline but the same behaviour occurs :(
More investigations on way ;)

lurch, 2016-03-08:
> it does not work displaying a for ever "waiting for settings partition".

Right, I guess that must be because of the partition-layout changes in NOOBS since I did my force-reinstall experimentation.

> I even tried to add runinstall at recovery.cmdline but the same behaviour occurs :(

Try `runinstaller

fthiebolt, 2016-03-08:
Here is my recovery.cmdline:
`runinstaller quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline kbdlayout=fr runinstaller silentinstall`
... so yes, i already added the `runinstaller` option to cmdline ... anyway it is still looking for the SETTINGS partition :|
I have the feeling that adding or removing something from the SETTINGS partition to force runinstaller to run would be a better option.

Regarding the `forcetrigger` option, the problem is that it is a permanent option (according to the doc).
Cheers

fthiebolt, 2016-03-11:
oh ... my mistake ... you're right Andrew, i just need to add `runinstaller` at the BEGINNING of the recovery.cmdline file :| 
I now have a fully automated solution to remotely install all of my Raspberry Pi directly from network and i can also tell you that it works with Raspberry Pi 3 :)

Cheers

lurch, 2016-03-11:
Sounds good :)

Cgettys, 2016-06-22:
Nice work! This is exactly what I need for a project I'm working on. @fthiebolt Do you have an example configuration file for installing a custom OS from network that I could see?

fthiebolt, 2018-11-08:
@procount hello, i'm currently giving a try to PiNN to automatically install an OS through the network. I downloaded pinn-lite and modified recovery.cmdline like this:

runinstaller quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline alt_image_source=http://xxxxxxxxxxxx/images/os_list.json keyboard=fr silentinstall select="Raspbian,waitnetwork" no_default_source no_update nofirmware ssh lang=en

RPi boot, i can see the 'Raspbian' from our server in the PiNN menu but it won't automatically start its installation ... missing option ?? :) Thanks

procount, 2018-11-08:
This is more of a PINN issue than NOOBs, so I've copied it to my repo https://github.com/procount/pinn/issues/196 . We should continue discussion over there.

procount, 2018-11-09:
@fthiebolt - please can you test the resolution in the link above

fthiebolt, 2018-11-12:
Hello,

I'll try by the beginning of the afternoon :)

See you later
François

procount, 2018-11-14:
Close?

fthiebolt, 2018-11-14:
Yep, things are working like a charm :)

Thanks for your help!
Best regards
François

XECDesign, 2018-11-15:
This one seems to be a wontfix in NOOBS. People who would like this feature, please use PINN.

closed 2018-11-15 by XECDesign